fieldOfView/Cura-OctoPrintPlugin #97: [Suggestion] Wait between gcode upload finishes and Starting a print.

Thisismydigitalself, 2018-12-12:
for the last couple of months or so i started using PTG plugin for octoprint which is far more accurate in predicting Print time. Cura ETA doen not take into considuration Printer's EEPROM settings and PTG calculate and factor these in so i am very happy with the results which are few minutes off for few hours print whereas Cura ETA is sometimes off by an hour - %% wise this is useless.
PTG plugin needs few seconds to do its thingy and calculate and it stops the moment Print starts, hence my following uggestions. I hope this can be added to Cura as I did not get an answer from PTG side.

Application Version
3.6.0

Platform
Win10/64/NV/SSD

Printer
CFFFP

Steps to Reproduce
upload gcode to printer using Cura / OctoPrint API 

Actual Results
When "Automatically start print" is checked i want my printer to wait a given seconds before print commences. i don't know if Cura's OP plugin can do this, or this is on Octoprint side to take of this.

Expected results
would like to have the option to chose how many seconds between uploading is down till print starts.

fieldOfView, 2019-01-24:
I think this should be handled on the OctoPrint side of things, but perhaps I can come up with something smart that is not ```if user=="Thisismydigitalself": do something```; I would prefer the plugin not becoming spaghetti code with all sorts of special exceptions for exotic usecases

fieldOfView, 2019-05-27:
I'm closing this because I think it should be handled on the server, either with a custom plugin for OctoPrint or in Print Time Genius.

fieldOfView, 2020-01-12:
I'm reopening this, while I'm trying to figure out if it is possible to check if analysis has been completed. I can't make any promises though.

fieldOfView, 2020-01-17:
This is now implemented. Could you test this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-17T14%3A22%3A37Z.curapackage?Expires=1894630961&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=gTC9Y69gIEJV1vIe7mBMjCL49TnBrtFmOtSTiKUC%2F7cLgoAxnVclLrancu%2BU2nK6L9eBb1ZhV2E3fPf0W90vQ9DIN0NR3IBxENrLSah8wI9kh2Xi%2F%2FPpYOttfh4PL0m9qTT3so4yPdryT%2F6uKOVaCO6lV6UFyhgIKBqNrowEd%2BJrunAbyBFZfix7QJjPJRTavVoL2AWR%2BOAVBXuL0qejJSNnebA8mpSeyJL3rLSFyofMgQaD%2F8OirSVVhWQ3deSCmZKTltKoHPxmN1r48uq6M0u%2BA1be4mp1UG8M1pbkr%2BuKIlWEhr9dLjGiTo0olWJI3H3uqS3VE6qwVQFoRXBAwA%3D%3D)?

commit e07208fa (2020-01-17): Wait for PrintTimeGenius gcode analysis before starting a print

Fixes #97
closed 2020-01-17 by fieldOfView