fieldOfView/Cura-OctoPrintPlugin #98: Plugin unchecks in Cura-->Marketplace-->Installed

matmis, 2018-12-14:
Everytime I check "OctoPrint Connection" in Cura-->Marketplace-->Installed it tells me that I need to restart Cura. But after restart the "OctoPrint Connection" plugin is unchecked again.

I'm using Cura version 3.6.0 on Arch Linux.

matmis, 2018-12-14:
I tried deleting `~/.config/cura`, but that does not fix the problem.

probonopd, 2018-12-14:
Run Cura from the command line and watch for errors.

matmis, 2018-12-14:
`2018-12-14 19:31:41,765 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Import error loading module UM3NetworkPrinting
2018-12-14 19:31:41,765 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2018-12-14 19:31:41,765 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/site-packages/UM/PluginRegistry.py", line 547, in _findPlugin
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:     module = imp.load_module(plugin_id, file, path, desc) #type: ignore #MyPy gets the wrong output type from imp.find_module for some reason.
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/imp.py", line 244, in load_module
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:     return load_package(name, filename)
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/imp.py", line 216, in load_package
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:     return _load(spec)
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 696, in _load
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
2018-12-14 19:31:41,766 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/__init__.py", line 4, in <module>
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:     from .src import DiscoverUM3Action
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/DiscoverUM3Action.py", line 17, in <module>
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:     from .UM3OutputDevicePlugin import UM3OutputDevicePlugin
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/UM3OutputDevicePlugin.py", line 15, in <module>
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]:     from zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo
2018-12-14 19:31:41,767 - ERROR - [MainThread] UM.Logger.logException [88]: ModuleNotFoundError: No module named 'zeroconf'
2018-12-14 19:31:41,768 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [587]: Could not find plugin UM3NetworkPrinting
2018-12-14 19:31:41,772 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Import error loading module UM3NetworkPrinting
2018-12-14 19:31:41,772 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2018-12-14 19:31:41,772 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/site-packages/UM/PluginRegistry.py", line 547, in _findPlugin
2018-12-14 19:31:41,772 - ERROR - [MainThread] UM.Logger.logException [88]:     module = imp.load_module(plugin_id, file, path, desc) #type: ignore #MyPy gets the wrong output type from imp.find_module for some reason.
2018-12-14 19:31:41,772 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/imp.py", line 244, in load_module
2018-12-14 19:31:41,772 - ERROR - [MainThread] UM.Logger.logException [88]:     return load_package(name, filename)
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/python3.7/imp.py", line 216, in load_package
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:     return _load(spec)
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 696, in _load
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/__init__.py", line 4, in <module>
2018-12-14 19:31:41,773 - ERROR - [MainThread] UM.Logger.logException [88]:     from .src import DiscoverUM3Action
2018-12-14 19:31:41,774 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/DiscoverUM3Action.py", line 17, in <module>
2018-12-14 19:31:41,774 - ERROR - [MainThread] UM.Logger.logException [88]:     from .UM3OutputDevicePlugin import UM3OutputDevicePlugin
2018-12-14 19:31:41,774 - ERROR - [MainThread] UM.Logger.logException [88]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/UM3OutputDevicePlugin.py", line 15, in <module>
2018-12-14 19:31:41,774 - ERROR - [MainThread] UM.Logger.logException [88]:     from zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo
2018-12-14 19:31:41,774 - ERROR - [MainThread] UM.Logger.logException [88]: ModuleNotFoundError: No module named 'zeroconf'`

I installed `python-zeroconf` and got it working.

closed 2018-12-14 by matmis

probonopd, 2018-12-14:
Are you using the AppImage?

matmis, 2018-12-15:
No, I installed Cura with Pamac.

fieldOfView, 2018-12-15:
Could you log an issue with whoever maintains the Cura package in Arch about the missing zeroconf?

The AppImage is the only officially supported way to run Cura on Linux. Any other ways are provided by third parties.

matmis, 2018-12-15:
I see it's included as an optional dependency.

[image]

dfredell, 2020-02-17:
I was getting the same error in my ~/.local/share/cura/4.4/cura.log file. And the OctoPrint plugin was disabled in the cura UI.

I too am on Arch Linux with cura 4.4.1 installed via `yay`
```log
 Could not find plugin OctoPrintPlugin
 Import error loading module OctoPrintPlugin

   File "/usr/lib/python3.8/site-packages/UM/PluginRegistry.py", line 540, in _findPlugin
 ignore #MyPy gets the wrong output type from imp.find_module for some reason.
   File "/usr/lib/python3.8/imp.py", line 244, in load_module
     return load_package(name, filename)
   File "/usr/lib/python3.8/imp.py", line 216, in load_package
     return _load(spec)
   File "<frozen importlib._bootstrap>", line 702, in _load
   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
   File "/home/danfredell/.local/share/cura/4.4/plugins/OctoPrintPlugin/OctoPrintPlugin/__init__.py", line 6, in <module>
     from . import OctoPrintOutputDevicePlugin
   File "/home/danfredell/.local/share/cura/4.4/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevicePlugin.py", line 7, in <module>
     from .zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo
   File "/home/danfredell/.local/share/cura/4.4/plugins/OctoPrintPlugin/OctoPrintPlugin/zeroconf.py", line 37, in <module>
     import netifaces
 No module named 'netifaces'

```

then later in the log was this
```
 Import error loading module UM3NetworkPrinting

   File "/usr/lib/python3.8/site-packages/UM/PluginRegistry.py", line 540, in _findPlugin
 ignore #MyPy gets the wrong output type from imp.find_module for some reason.
   File "/usr/lib/python3.8/imp.py", line 244, in load_module
     return load_package(name, filename)
   File "/usr/lib/python3.8/imp.py", line 216, in load_package
     return _load(spec)
   File "<frozen importlib._bootstrap>", line 702, in _load
   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
   File "/usr/lib/cura/plugins/UM3NetworkPrinting/__init__.py", line 3, in <module>
     from .src import UM3OutputDevicePlugin
   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/UM3OutputDevicePlugin.py", line 12, in <module>
     from .Network.LocalClusterOutputDeviceManager import LocalClusterOutputDeviceManager
   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/LocalClusterOutputDeviceManager.py", line 15, in <module>
     from .ZeroConfClient import ZeroConfClient
   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 8, in <module>
     from zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo
 No module named 'zeroconf'
 Import error loading module AMFReader
```

I too solve this issue by installing the `community/python-zeroconf` package via `yay`

fieldOfView, 2020-02-17:
> I too am on Arch Linux with cura 4.4.1 installed via `yay`

Then whoever maintains the Cura package for Arch Linux needs to fix the dependencies of that package. `netifaces` needs to be added. It also looks like `trimesh` is missing which causes other modules and plugins to fail to load (judging by the import error in AMFReader).

The recommended way to run Cura on any Linux flavor is to use the AppImage provided by Ultimaker.