fieldOfView/Cura-OctoPrintPlugin #100: Webcam image ignores flip-settings

muesli, 2019-01-11:
I have to run my Pi cam upside down for cable management reasons. Luckily there's a setting in OctoPrint to flip the image vertically/horizontally. It would be nice if Cura's OctoPrint plugin could respect that setting when displaying the stream.

Should this be a bug in OctoPrint instead, please let me know and I'm happy to file a new ticket there.

Thanks for the great work!

fieldOfView, 2019-01-11:
What version of Cura, and what version of the plugin do you use?

muesli, 2019-01-11:
Cura 3.6.0 and plugin from git. I've tried installing it from the marketplace first, but that failed on ArchLinux.

fieldOfView, 2019-01-11:
I run an upside-down camera myself, and it displays correctly in Cura with OctoPrintPlugin.

Did you get the 3.5 branch, and is it up-to-date? What version do you have, according to plugin.json?

fieldOfView, 2019-01-12:
Don't use the master branch. Use the 3.5 branch instead for Cura 3.6.

muesli, 2019-01-14:
Ok, got it working. Using the 3.5 branch seemed to do the trick - either that or I simply had to restart OctoPi after changing the setting. Thanks for your help!

closed 2019-01-14 by muesli