fieldOfView/Cura-OctoPrintPlugin #101: Octoprint webcam not visible in Cura 3.6

gkrystev, 2019-01-15:
Similar with issue #76, the webcam on monitor tab do not show, or it shows after 30-60 seconds, with no live picture but only still images changing time to time (about each 60 seconds).
I did try to fix the issue with the script (OctoPrintOutputDevice.py) proposed on [Ultimaker/Cura#3725](https://github.com/Ultimaker/Cura/issues/3725) but with that new script, the monitor tab shows only that the printer is not connected.  With the original script everything but the webcam works just fine. 
Plugin version is 3.5.3
Cura version is 3.6.0 (running on Windows)

fieldOfView, 2019-01-16:
What is the resolution of the image, and do you connect through wifi or network cable?

fieldOfView, 2019-01-16:
Replacing files with versions of the plugin for different versions of Cura is never a good idea; the Cura issue you linked to was about Cura 3.4, so you were trying to mix a file for Cura 3.4 with a plugin for Cura 3.6.

gkrystev, 2019-01-16:
The OctoPrint is connected with cable, the Cura is running on laptop connected via WiFi
Image resolution is 640x480. The stream is working fine from OctoPrint web interface and from the direct link.

fieldOfView, 2019-01-16:
Then I'm going to need logs. 
* Go to Help -> Show configuration folder.
* Locate Cura.log
* Save it to dropbox, google drive, wetransfer, or a service like that and post a link here.

gkrystev, 2019-01-16:
Logs: https://pastebin.com/2cLUtkz4

fieldOfView, 2019-01-16:
It looks like the plugin cannot parse the MJPEG stream; it cannot find the "end of jpeg" marker in each frame. What usb camera do you use? Can you try a different camera?

gkrystev, 2019-01-17:
in OctoPrint I'm using mjpg_streamer for the streaming. The camera is some Logitech HD Pro (C930 I think). The settings for the mjpg_streamer are: -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 640x480 -f 10

Should a webcam model make a difference? As I think after the stream is created the output should be the same, only the image quality will differ.

fieldOfView, 2019-01-17:
Yes, the webcam model can make a difference. If the webcam natively supports mjpeg, mjpeg-streamer does not reencode that stream but just passes it along.

Cura contains a somewhat naive mjpeg parser that picks the stream apart and displays it frame-by-frame. All the mjpeg streams I have seen sofar are jpeg images concatenated one after the other. Yours seems (at least slightly) different.

fieldOfView, 2019-01-18:
To help me better support your webcam, would you be so kind to store a couple of seconds of your webcam stream? It is not entirely trivial to save an MJPEG stream (because it is basically sent as an infinitely large file), but I have found a way using VLC.

* Create an empty file named eg "dump.mjpg"
* Open VLC, and go to Media -> Convert / Save
* On the Network tab, enter your stream URL and press Convert/Save
* In Settings, change from "Convert" to "Dump raw input"
* In Destination, select the empty file (dump.mjpg) to overwrite
* Press start
* Count to 5 and close VLC

The dump.jpg should now have considerable content. Please zip that up and publish it somewhere.

gkrystev, 2019-01-21:
I did able to capture something.
Still, when I play the video in MPC for example, the video speed seems about 2 times faster than actual speed.
Here is a link to the raw video: http://s000.tinyupload.com/index.php?file_id=38211431375667387032

fieldOfView, 2019-01-21:
Awesome, thanks. I’ll have a look tomorrow.

The playback speed thimg is to be expected; there is no timing info in the stream so it just plays “as quick as possible”.

fieldOfView, 2019-01-22:
Ok, sofar for that theory; the mjpeg stream seems perfectly valid. That leaves me a bit at a loss as to why this is happening.

The logs show that the buffer that is used to decode the mjpeg stream fills up up > 2MB in ~ 20 seconds, all while unable to find the start and/or end frame markers which are clearly there in the stream. All I can think of is that the stream arrives in chunks of ~2 MB at a time, or something like that.

I did notice that you are not using the standard reverse proxy settings (port 88 instead of port 80). That by itself should not make a difference, but perhaps you made other changes too? I'm grasping at straws here...

gkrystev, 2019-01-22:
The stream is behind HAProxy, not directly from mjpg_streamer
HAProxy config is trivial

frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/server.pem
        option forwardfor except 127.0.0.1
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server localhost 127.0.0.1:8080

Response headers that I'm getting from the stream are:
HTTP/1.0 200 OK
Access-Control-Allow-Origin: *
Server: MJPG-Streamer/0.2
Cache-Control: no-store, no-cache, must-revalidate, pre-check=0, post-check=0, max-age=0
Pragma: no-cache
Expires: Mon, 3 Jan 2000 12:34:56 GMT
Content-Type: multipart/x-mixed-replace;boundary=boundarydonotcross

Not sure how to set any chink size or so.

If Cura has naive mjpeg parser as you said, what is changed in this parser from version 3.5, as in the older Cura version the stream works like a charm.

fieldOfView, 2019-01-23:
>  in the older Cura version the stream works like a charm

That would have been worth mentioning before...

fieldOfView, 2019-01-24:
Between Cura 3.5 and Cura 3.6, a memory leak in showing the webcam stream was fixed. The memory leak was introduced by updating the GUI framework Qt in Cura 3.5. The component that show the MJPEG stream was refactored and should even be performing slightly better. Apparently, not for everybody. 

I am still not closer to finding out what happens on your side.

fieldOfView, 2019-01-28:
Could you test http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2019-01-28T14_45_08Z.curapackage ? Drop the file in a running Cura window and restart Cura.

While it does not directly change the mjpeg code, there's a small chance it fixes the issue. And it would be good to know if this version makes things worse.

gkrystev, 2019-02-01:
Sorry for the late response.
Out of the blue, the camera/monitoring start working again.
I did not change any files or update new versions of Cura or Octoprint plugin. Still, the camera works fine.
I did restart the printer, the OctoPi, Cura and my computer several times with no luck, but not all works fine...

fieldOfView, 2019-02-02:
Always an awkward outcome, if problems go away automagically; they might come back automagically too...

fieldOfView, 2019-02-02:
I'm closing this for now, let me know if the issue resurfaces. I'll frown some more then....

closed 2019-02-02 by fieldOfView