fieldOfView/Cura-OctoPrintPlugin #102: Octoprint entry loses focus when attempting to edit entry.

chezhalcyonacres, 2019-01-23:
Cura 3.6
Plugin 3.5.3
Windows 10/64
ref issue #95

fieldOfView, 2019-01-24:
I am going to need more info than this. Please add steps to reproduce, or a link to a video showing what happens.

chezhalcyonacres, 2019-01-24:
I recorded my session here: https://www.youtube.com/watch?v=ty0YxtyY_Kg
Cura crashed and closed at the end.

fieldOfView, 2019-01-25:
That looks annoying. It is still not quite clear to me what happens: when you select an instance from the list, does it get deselected automatically or when you (try to) click the field or button?

Did previous versions of the plugin work correctly?

Could you upload your cura.log to pastebin.com or similar (see Help -> Show configuration folder) and post a link here?

chezhalcyonacres, 2019-01-25:
https://pastebin.com/SzgPAxLD

What happens is the top entry stays grey highlighted, and reverts to that state after clicking any of the entries. I showed my attempts to get the API key entered, but if I do nothing, it reverts to the same. 

This is a new install, so no prior version was tested.

fieldOfView, 2019-01-26:
Looks like this may be caused by my attempt to restart bonjour discovery when it crashes:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/4894ae06460cbd038102348db768ecc28ea031d1

chezhalcyonacres, 2019-01-26:
My bonjour instance looks pretty old (2015), even though iTunes just updated. Is there a direct way to get it  updated?

chezhalcyonacres, 2019-01-26:
OK, I fixed it. 

It seems I had inbound traffic blocked on windows firewall for Cura. 
I enabled both directions and now it's working as expected. 

Sorry for raising this as a bug, hopefully someone else can benefit,

fieldOfView, 2019-01-26:
Great, thanks. OctoPrintPlugin should still behave better though.

fieldOfView, 2019-01-28:
Could you test http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2019-01-28T14_45_08Z.curapackage ? Drop the file in a running Cura window and restart Cura.

chezhalcyonacres, 2019-01-30:
> Could you test http://files.fieldofview.com/cura/OctoPrintPlugin-v5.0.0-2019-01-28T14_45_08Z.curapackage ? Drop the file in a running Cura window and restart Cura.

The plugin behaves much better now. 
I set the firewall back to "outgoing only" and was able to manage the OctoPrint entry.
Cura still needs to be able to accept connections though or it won't slice, so had t set it back to "allow all"

Thanks!

fieldOfView, 2019-01-31:
Good to know this version behaves better. I have submitted it to the Marketplace.

> Cura still needs to be able to accept connections though or it won't slice

Off-topic, but Cura should only need to be able to accept _local_ connections.

closed 2019-01-31 by fieldOfView

chezhalcyonacres, 2019-01-31:
yeah, it creates a socket listener to connect to Octo. 

`2019-01-31 13:45:29,256 - DEBUG - [Thread-12] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Could not connect to the given address
2019-01-31 13:45:29,268 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [809]: Backend quit with return code 0. Resetting process and socket.`