fieldOfView/Cura-OctoPrintPlugin #104: Print location dropdown menu at the wrong place on mac

calania, 2019-02-18:
Hello, I am having an issue with this plugin running on cura 3.6 on a mac where if you press the dropdown menu next to the "print with octoprint" button in the bottom right corner to select "save to file" instead the box is located in the upper right corner, far from where it should be (See picture) [https://i.imgur.com/Sit3BAq.png](url) This might be a bug in cura and not as a result of your plugin but just wanted to report it to you in hopes that it might be fixed.

fieldOfView, 2019-02-18:
It is indeed a bug in Cura. Nothing I can do to fix it from the plugin. It should be fixed in the upcoming Cura 4 beta 2.

closed 2019-02-18 by fieldOfView

calania, 2019-02-18:
Sound's good, thanks for the fast reply!