fieldOfView/Cura-OctoPrintPlugin #106: Cura not reconnecting after being taken off network

begberts, 2019-03-15:
I have Cura installed on my laptop and Octoprint running for my printer on my home network. I bring my laptop to work and other places most days and when I bring it back home Cura will not reconnect to Octoprint. The only way I can reconnect it is by restarting the laptop. 

This is especially annoying when I VPN into my network and try to print remotely. I can't seem to get that to work at all but that may be an different issue. 

My guess is that the plugin (or Cura) isn't trying to reconnect each time it opens.

fieldOfView, 2019-03-15:
What version of the plugin is installed?

When you go into the Connect to OctoPrint dialog, and you press the Refresh button there, does the OctoPrint instance show up?

begberts, 2019-03-15:
Version 3.5.4 on Cura 3.6

I just VPNed in and the Octoprint instance is not showing up (probably a different issue?). If I recall correctly the same thing happens when I get home and reconnect my laptop to my home network. Cura just doesn't show the connection. 

I can confirm later today when I get home. 

I honestly don't know if this is an issue with the plugin or my Octoprint/network configuration. I've played around with the setting and reset everything to defaults but have still had the same issue. I just don't know what else it could be. I'm pretty new to networking and working with Pi's.

fieldOfView, 2019-03-15:
When you go into the Connect to OctoPrint dialog, and you press the "Refresh" button there, does the OctoPrint instance show up in the list?

Can you reach the OctoPrint instance through its name (eg octopi.local) or ip address? If so, have you tried manually adding the OctoPrint instance using the Add button?

begberts, 2019-03-15:
I did below while connected via VPN to my home network. 

The instance did not show up after refreshing the Connect to OctoPrint dialog. 

I am able to ping and ssh into the pi as normal.

I added the instance manually and it appears to be working as expected when I VPN into my network.

I can confirm that it automatically connects once I get home but I don't think that will be an issue anymore. 

Thanks for helping troubleshoot this. I guess it wasn't really an issue after all.

closed 2019-03-20 by begberts