fieldOfView/Cura-OctoPrintPlugin #107: G-code not converted to X3G before sending

Ghostkeeper, 2019-03-20:
I got a bug report [here](https://github.com/Ghostkeeper/X3GWriter/issues/25) that my X3GWriter plug-in isn't working when sending to Octoprint. I looked through the source code to see what happened and found that the Octoprint plug-in is [always calling the g-code writer](https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/6949f1eb429021f632bc560d494812cf41087c51/OctoPrintOutputDevice.py#L329). So the conversion is never done for printers that require a different format than g-code.

There is a workaround for this issue, because there is a server-side [plug-in for Octoprint](https://plugins.octoprint.org/plugins/gpx/) that does the same thing. But I guess it's a bit easier to be able to use X3G via Cura since Cura knows the machine settings to do the conversion with.

It does raise the question though, what happens if you start sending (binary) X3G data to Octoprint? Does it support that at all?

fieldOfView, 2019-03-20:
The reason OctoPrint does this is that it is the gcodewriter that adds the (encoded) settings summary at the end. I'll look into an alternative. Note that since recently, USBPrinterOutputDevice seems to do the same thing.

Ghostkeeper, 2019-03-21:
I think that calling the g-code writer to produce the g-code is a good thing. It adds the settings summary and processes the post-processing scripts as well.

In the issue in the X3GWriter repository you said that OctoPrint only supports g-code data, not X3G. So then calling on the g-code writer makes sense and this is not really an issue. Same goes for USBPrinterOutputDevice. For other devices such as RemovableDriveOutputDevice it should use the preferred output format or allow the user to select it.

closed 2019-03-21 by Ghostkeeper