fieldOfView/Cura-OctoPrintPlugin #108: Web interface of Octoprint crashes

babylongithub0815, 2019-03-24:
Web interface of Octoprint can no longer be used when you press the button "house" in Cura, go to zero position.

Weboberfläche von Octoprint lässt sich nicht mehr bedienen wenn man in Cura auf den Button "Haus" drückt, für gehe in Null Position.

fieldOfView, 2019-03-25:
Can you be a little more specific? What do you mean by "can no longer be used"? What happens when you try? 

Does the same button in the OctoPrint web interface work as expected?

What printer do you use?

incrediblesolutions, 2019-05-17:
having a similar issue with a fresh install.  going to reinstall as finding documentation of the code seems to be difficult.  i would like to find a wikki or doc web site that has full documentation of what the Octopi Part is

fieldOfView, 2019-05-17:
Before you ask documentation from me, might I be so bold to ask you to actually describe your problem to me? All you say is that you have a "similar" problem to a problem where I said I need more information to be able to help. How do you expect me to know what is going wrong?

fieldOfView, 2019-05-27:
Closing due to lack of information.

closed 2019-05-27 by fieldOfView