fieldOfView/Cura-OctoPrintPlugin #109: Cura Crash to Desktop

Schriazugul, 2019-03-25:
If i try to open the "Monitor" tab within cura it crashes to desktop all the time.
i am not able to print anymore

this is happening since i updated to the latest nvidia driver (opengl v. 4.1.0)

**Further information:** 
[image]

**Cura Logs:**
Thread 0x00002108 (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000501c (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 161 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000aa8 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Op3r4t0r\AppData\Roaming\cura\4.0\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1725 in wait
  File "C:\Users\Op3r4t0r\AppData\Roaming\cura\4.0\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1326 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000c14 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Op3r4t0r\AppData\Roaming\cura\4.0\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1725 in wait
  File "C:\Users\Op3r4t0r\AppData\Roaming\cura\4.0\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1190 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000e58 (most recent call first):
  File "C:\Users\Op3r4t0r\AppData\Roaming\cura\4.0\plugins\OctoPrintPlugin\OctoPrintPlugin\zeroconf.py", line 1111 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000302c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000c98 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00005654 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003af4 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003db4 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\UM3NetworkPrinting\src\UM3OutputDevicePlugin.py", line 344 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000058dc (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001fcc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00005160 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002ad4 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000f84 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003dc4 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000053a8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004e1c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000003f4 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000028a0 (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "X:\4.0\build\inst\bin\cura_app.py", line 98 in exceptHook
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\ControllerProxy.py", line 104 in _onActiveStageChanged
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315 in __performEmit
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212 in emit
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Controller.py", line 166 in setActiveStage
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\ControllerProxy.py", line 47 in setActiveStage
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 168 in mouseReleaseEvent
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 324 in exec_
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 761 in run
  File "X:\4.0\build\inst\bin\cura_app.py", line 132 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Schriazugul, 2019-03-26:
additional note: 
downgrading the driver had no effect. 
Still crashes with the same error

fieldOfView, 2019-03-26:
Please share cura.log from Help -> Show configuration folder

Schriazugul, 2019-03-26:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3010731/cura.log)

fieldOfView, 2019-03-27:
Thanks. That did not make it clearer why this happens :-/ At the moment it looks like the crash doesn't have much to do with the OctoPrint plugin itself, but I'll still try to find the cause of the crash.

This is just a hunch: could you try right-clicking the viewport (to open a context menu) before switching to the Monitor?

nallath, 2019-03-27:
I don't think this is a Octoprint plugin issue, but I also can't tell what is causing the issue then. It seems some kind of race issue (which suddenly popped up because you upgraded your drivers) that is caused by the garbage collection of python.

Schriazugul, 2019-03-27:
Thank you Guys for investigating, i do really appreciate it. 

> This is just a hunch: could you try right-clicking the viewport (to open a context menu) before switching to the Monitor?

i will test this again this evening but am fairly sure that the crash also happens than. 

i can use cura itself pretty normally. 
- Import stl works
- change options works
- slicing works
- preview tab is working
- i can even use other plugins

therefore i tried: 
- downgrading the driver -> still crashing
- removing the ocotoprint pluging -> can use monitor tab, no crashes
- reinstalled octoprint plugin -> crashes happening again.
- installed cura 3.6 with octoprint plugin -> also crashes

because of that i created the issue here, to me it looks like octoprint plugin was causing the crashes.

are there any plugin based settings which i can remove to kinda reset the plugin?

fieldOfView, 2019-03-27:
Thanks for testing. 

We think it is a problem in Cura which is triggered by the OctoPrint plugin. The crash happens outside the plugin. For you that makes no difference; you just need it fixed. "Unfortunately" the crash does not happen on my end, which makes it harder to figure out what is going on.

The right mouse button thing is just a last straw. Something gets initialised the first time you open the context menu of the viewport, which may be left unitialised otherwise.

To remove all settings made by the OctoPrint plugin, go to Help -> Show configuration folder. 
Close Cura
Open `cura.cfg` in a text editor, and remove the [`octoprint`] section (there may be a `manual_instances` setting in it) 
Open the `machine_instances` folder, and for each *.global.cfg file, open it in a text editor and remove any line starting with `octoprint_` (`octoprint_id = ...` and `octoprint_api_key =...`)

Schriazugul, 2019-03-27:
ok i tried the right klick thing but unfortunately it doesnt help.
afterwards i removed all of the configuration as you mentioned, loaded up cura, octoprint plugin still installed - > was able to open the monitor tab. 
afer i configured the octoprint connection again within the printer settings, it again crashes. 

i also tried to not use zeroconf and set the connection to static but this doesnt help too

so i ended up with the idea of testing another windows and cura installation on a notebook
(no worries i got your point that cura throws the error but i want to do something about it.. )
turns out that: 
- untouched but up to date windows installation with freshly installed cura + plugin
-> cura is crashing with the same error. 

so actually it drives me crazy because i have not changed my octoprint setup and the cura installation is totally default.

Schriazugul, 2019-03-31:
latest discoveries are: 
- freshly installed octoprint (octopi 16.0) without any plugin: still crashes on both computers
- it is not crashing if i do not connect octoprint to the printer (serial connection)

i think i'll now give up on this, considering cura as to buggy to use. 
i have no idea why it stoped working i even replaced every side which can cause the issue execpt the printer itself. looks like i have to go back to slicer pe. 

anyways thank you guys for looking at this and all of your help. 
i will keep cura installed, if you need more information from me just let me know.

fieldOfView, 2019-03-31:
Thanks for your continued testing. The second point - not crashing when there is no serial connection to the printer - gives us some further points to investigate, though not yet being able to reproduce the issue does not make fixing it any easier.

Don't discount Cura on behalf of the connection with OctoPrint not working ideally though. It is a good slicer.

edwardteach42, 2019-04-05:
I am experiencing this exact same problem only in Cura 4 and it only happens when I click "connect" octoprint.  Works fine in 3.6.

**Error Traceback**
```
Traceback (most recent call last):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\OctoPrintPlugin\DiscoverOctoPrintAction.py", line 228, in setApiKey
    self._network_plugin.reCheckConnections()
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\OctoPrintPlugin\OctoPrintOutputDevicePlugin.py", line 145, in reCheckConnections
    self._instances[key].connect()
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\OctoPrintPlugin\OctoPrintOutputDevice.py", line 254, in connect
    self.setConnectionState(ConnectionState.connecting)
  File "enum.py", line 274, in __getattr__
AttributeError: connecting
```

**Logs**
```
Thread 0x000055ac (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00005734 (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 161 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003894 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001f50 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000f94 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00007cac (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00008690 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\UM3NetworkPrinting\src\UM3OutputDevicePlugin.py", line 344 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00005ff0 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001ccc (most recent call first):
  File "threading.py", line 297 in wait
[... 76 lines cut ...]
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x000073e0 (most recent call first):
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "X:\4.0\build\inst\bin\cura_app.py", line 98 in exceptHook
  File "X:\4.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 324 in exec_
  File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 761 in run
  File "X:\4.0\build\inst\bin\cura_app.py", line 132 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

```

fieldOfView, 2019-04-05:
@edwardteach42 you are experiencing a different problem. Yours is more easily fixable: In the Marketplace, uninstall the OctoPrint Connection plugin, restart Cura, go back into MarketPlace and install the plugin again. Finally restart Cura for a second time.

The line that crashes is not in the current version on the Marketplace. I don't know what exact version you are using, but it is not a version that has been officially distributed for Cura 4.0 in the Marketplace.

edwardteach42, 2019-04-10:
@fieldOfView that worked. I also had to delete the octoprint folder in the plugins after uninstalling.

Schriazugul, 2019-04-29:
i wanted to give you a short update on this. 
looks like i found the issue and it is kinda unexpected. 
My Pi is connected to the local network with wifi and the connection is very poor most of the time but it was enough to upload files to the printer. 
I did not ever use wifi so i gave it a try and used cabled network again and it worked immediatly! 
than i changed the connection back to wifi, it crashed again.. 
so looks like that cura or the octoprint plugin is not very fault tolerant when it comes to package loss due to bad (wifi) connectivity.  
since then i reloacted the printer to achieve better connection and now i am able to use cura with octoprint plugin most of the time, it is still crashing from time to time but it does always complete the upload before it finally dies which suits my needs ^^

i hope this information does help in someway..

fieldOfView, 2019-06-26:
I have not (entirely) forgotten about this issue, but I am also no closer to finding out what is going on. The network quality affecting the issue is an interesting observation, but I have not yet been able to reproduce the issue.

fieldOfView, 2019-08-12:
Is this still an issue with the latest version?

Schriazugul, 2019-08-13:
since i switched back to cabled connection i did not have issues. (cura and plugin are ofc updated)
i will do some testing this week while my rapsi is using the wifi.

fieldOfView, 2019-08-13:
Thanks, that would be great