fieldOfView/Cura-OctoPrintPlugin #112: [4.0] Octoprint camera picture in Monitor window not showing with reverse proxy that requires http auth with username and password #5546

sjbotha, 2019-04-01:
Application Version
4.0 and 3.6

Platform
Windows 10

Printer
Tevo Tornado

Steps to Reproduce
Run Cura and click Monitor

Actual Results
No webcam image

Expected results
The webcam image is shown

Additional Information
I'm using Cura 4.0 right now. I also had this issue with Cura 3.6.

My octoprint is accessible at /octoprint path and webcam is at /octoprint/webcam. After configuring the reverse proxy to not require authentication for /octoprint/webcam the image showed up.

I made logs of the camera not working and then working. I did not see a difference in the log.

[cura webcam not working.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3029684/cura.webcam.not.working.log)
[cura webcam working.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3029685/cura.webcam.working.log)

fieldOfView, 2019-04-01:
> After configuring the reverse proxy to not require authentication for /octoprint/webcam the image showed up.

Does your webcam require the same username/password as the octoprint interface?

sjbotha, 2019-04-01:
Yes. It is connected directly to octoprint and it's under the same
/octoprint path so I just forward both together from the reverse proxy
server and it requires auth for everything under /octoprint. I guess you
could also have a separate IP camera setup on a different host with a
different IP and then you setup a separate reverse proxy config with no
auth.

fieldOfView, 2020-01-13:
Fixed in 221098c, sorry it took a while.

commit 221098c2 (2020-01-13): Let camera url share basic auth with octoprint url

Fixes #112
closed 2020-01-13 by fieldOfView