fieldOfView/Cura-OctoPrintPlugin #113: Need the ability to upload a file, but not have it selected for printing.

jbernardis, 2019-04-30:
I often upload files to octoprint while it is actively printing.  They are simply stored in the Pi's memory, and they are selected for print later, either through the octoprint GUI or through a different client.  This essentially creates a print queue.

streaky, 2019-05-26:
+1 my workflow tends to be to stack things up to print later. What I'm having to do with the plugin is cancel the print whilst it's preheating which means waiting for it to do that. It works for casual but if you have a bunch of stuff you want to queue up for later use the workflow is inflexible.

fieldOfView, 2019-05-27:
In the "Connect to OctoPrint" dialog, there is an checkbox labeled "Automatically start print job after uploading", which is selected by default. Does deselecting this work for you?

jbernardis, 2019-05-27:
No.  There is a difference between selecting to print and printing.  If I upload after deselecting this, the file is uploaded and selected such that I only need to press the print button.  I want the file to stop after the upload, waiting to be selected for printing by the client.

streaky, 2019-05-27:
It certainly _helps_, but it's not the same thing. Best solution IMHO would be to have two print menu items - assuming that's possible - one works as now, the other just simply uploads the file. Print With OctoPrint / Save to Octoprint.

fieldOfView, 2019-05-27:
@jbernardis I don't get what's the problem with having the uploaded file selected. Can you elaborate?

jbernardis, 2019-05-27:
What happens if the printer is actively printing?   Will the upload terminate the job to handle the new file?  Or will the upload fail?  Or will the upload succeed but the selection fail?  In any case, we are counting on a failure to get what we want.

What happens if I have a file on the pi that I want to print next?  Uploading forces the new file in front of the one I had in mind.

fieldOfView, 2019-05-27:
>  Will the upload terminate the job to handle the new file?
No.
> Or will the upload fail?
No.

Nothing will fail. If there is a file printing, you can still select another file in OctoPrint. You just can not start it, until the already printing file is either done or canceled by the user.

This is how queueing works in OctoPrint. The OctoPrint Connection plugin will not reimplement or work against functionality in OctoPrint.

jbernardis, 2019-05-27:
I just verified that the job will upload successfully and it will not interrupt a printing job, but you absolutely cannot select another file for printing while one is already printing - octoprint rejects the request. 

Fix it or not as you want.

fieldOfView, 2019-05-28:
It works as expected then.

closed 2019-05-28 by fieldOfView

jbernardis, 2019-05-28:
If expected means that you are hardcoding a parameter in the API  as true, then yes it works in the limited way that the author intended.  Why exposing this parameter is such a hardship I'll never know.

fieldOfView, 2020-02-28:
After some deliberation, this not automatically selecting the file is now the default behavior unless you set the option to start printing automaically.

Further discussion in #154