fieldOfView/Cura-OctoPrintPlugin #114: Make printer show up in Cura 4.1 "Add a neworked printer" dialog

probonopd, 2019-05-24:
It would be nice if OctoPrint printers would show up in the new Cura 4.1 "Add a neworked printer" dialog that appears when you open the app for the first time:

[image]

If Cura detects an OctoPrint printer, it would ideally suggest to download the OctoPrint plugin.

Since probably Cura functionality is required to implement this, I will open a feature request there as well.

fieldOfView, 2019-05-25:
A major problem here is that OctoPrint does not have a set way to identify the make/type of printer so you would still need to pick that from a list to get the appropriate settings and profiles in Cura. Provided OctoPrint has been properly configured, you can get the basics from the OctoPrint API (bed size and shape, nozzle size), but Cura printer definitions are much richer than that.

Deneteus, 2019-12-13:
I name all of my octoprint instances and I use static IP's with names so I can identify what printer it is I am trying to connect to so this would come in handy.

fieldOfView, 2020-01-14:
> I name all of my octoprint instances and ...

That does not help Cura to know what printer type they are. I still need to figure out a way to show a printer type dialog after selecting the OctoPrint instance.

probonopd, 2020-01-14:
What I find strikingly strange is that one typically has to enter things like the bed size, filament type,  nozzle size and other vital machine and consumables parameters into the _slicer_ (each time one slices something) rather than into the _printer_ (at the time time when one changes the filament, the nozzle, etc) which could then self-describe itself to the slicer. The "traditional" model of doing things makes sense if the printer is disconnected from the slicer (e.g., through the use of SD cards) but is not the best way to do things once the printer is connected to the slicer (by USB or Ethernet or wireless). Probably the printer firmware, print "spoolers" (like OctoPrint), and slicers would need to work hand-in-hand to achieve the best possible plug-and-play experience.

Sounds to me like someone should propose a standard for 3d printers describing themselves over networks for true plug-and-play operation. Or does this already exist? From a cursory glance it seems that the Internet Printing Protocol extension for 3D Printing might address this need: https://www.pwg.org/3d/ addresses thiss: https://ftp.pwg.org/pub/pwg/candidates/cs-ipp3d11-20190329-5100.21.pdf

fieldOfView, 2020-01-15:
Selecting a printer in Cura does much more than just setting printer dimensions. It also selects printer profiles made and tuned specifically for that printer and the materials it supports.

probonopd, 2020-01-15:
> Selecting a printer in Cura does much more than just setting printer dimensions. It also selects printer profiles made and tuned specifically for that printer and the materials it supports.

Exactly. I think that the IPP 3D Printing Extensions, by describing the printer and its loaded materials, could facilitate the auto-discovery and auto-configuration of both printers and materials, if implemented everywhere (firmware, print spooler, slicer). Or if it can't, it should be extended so that it can.