fieldOfView/Cura-OctoPrintPlugin #115: Information not refreshing with v3.5.5 in Cura 4.1

dnblankedelman, 2019-05-31:
Today I rev’d to 4.1 of Cura and at the same time updated to 3.5.5 as per the update suggestion under Marketplace in the software. My Octoprint install itself hasn’t changed.

I now find that the plugin has difficulty displaying updated information. for example, unless I quit and restart Cura, it doesn’t show an updated printing time or estimated completion time. Nor does it update the percentage complete indicator or the bed/nozzle temperatures (the webcam view works fine, probably because that’s just an embedded stream).

If I go back to using Cura 4.0 with 3.5.4, everything updates fine leading me to believe there’s an issue with one of the 3.5.5 commits (which I was made to specifically address Cura 4.1).

Is there something I can do to help troubleshoot this? 

Thanks!

fieldOfView, 2019-06-01:
Could you try if version 3.5.5 of the plugin works with Cura 4.0 for you? The changes in Cura 3.5.5 are very minor, I would not expect them to cause this.

dnblankedelman, 2019-06-02:
Didn't want to leave you hanging, but unfortunately I don't seem to be able to answer this question.

Each time when I attempt to upgrade (Start Cura 4.0, Marketplace -> Installed -> Upgrade), the upgrade from 3.5.4 to 3.5.5 appears to time
[curauplog.txt](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3245107/curauplog.txt)
[curauplog.txt](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3245110/curauplog.txt)

out and throws this traceback:

Traceback (most recent call last):
  File "/Applications/Ultimaker Cura4.app/Contents/MacOS/plugins/plugins/Toolbox/src/Toolbox.py", line 669, in _onDownloadProgress
    Logger.log("w", "Failed to download package. The following error was returned: %s", json.loads(bytes(self._download_reply.readAll()).decode("utf-8")))
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/json/__init__.py", line 319, in loads
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/json/decoder.py", line 339, in decode
  File "/Users/ultimaker/build/env/4.0/inst/lib/python3.5/json/decoder.py", line 357, in raw_decode
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

There's a considerably longer Logs output, I've attached it as a file.

dnblankedelman, 2019-06-02:
Yup, confirming that I'm being blocked by a server-side issue at the moment. In cura.log, it says:

> Attempting to download & install package from https://api.ultimaker.com/cura-packages/v1/cura/v6.0.0/packages/OctoPrintPlugin/download

which redirects to https://storage.googleapis.com/ultimaker-cura-packages… which is currently throwing 502 errors.

Standing by….

dnblankedelman, 2019-06-02:
Ok, server-side outage over, was able to install 3.5.5 plugin into 4.0.0 and…

same symptoms in 4.0.0 as 4.1.0. Monitor tab only refreshes if I quit and restart Cura.

fieldOfView, 2019-06-03:
Could you try manually adding the OctoPrint instance instead of using the automatically discovered one from the list? The only functional difference between the 3.5.5 and 3.5.4 release is in the zeroconf (bonjour) discovery.

dnblankedelman, 2019-06-03:
Sorry, I must not have been clear. I have only added it manually by using the hostname "octopi.local'. It never shows up as a choice to be added via a discovery mechanism. If you'd like, I can remove it and try using an IP address. I haven't done that for fear of the host chasing IPs if it loses its DHCP lease. 

As a related aside, since it connects fine and gets data when Cura is first launched, is there a difference between how the initial connect and subsequent refreshes are performed?

fieldOfView, 2019-06-03:
>  I can remove it and try using an IP address.

Please try. octopi.local may resolve to either an IPv4 or IPv6 address. In my experience IPv6 is often much slower.

dnblankedelman, 2019-06-03:
Ok, tried this in Cura 4.1.0 (removed the OctoPrint instance via 3.5.5, added it back using IP address instead of hostname). No difference in the symptoms.

dnblankedelman, 2019-06-03:
BTW, if it helps at all to pass along a log or to run a debugging version (anything I can do to help, really), please let me know. Really appreciate this plugin.

dnblankedelman, 2019-06-05:
So, here's an interesting piece of data: 

I blew away the plugin from the plugin folder. I cloned the repos, and then rolled it back to the commit right before you downgraded zeroconf (git reset --hard b8dc8134785124006f498865c9dc898bb5a99031). 

Plugin was back to its old self and working fine. 

@fieldOfView Any thoughts?  

Still happy to roll forward to current, then send you cura logs or whatever if that would be helpful.

fieldOfView, 2019-06-26:
Sorry for taking a while to get back to this. Could you try unchecking the "Automatically discover local OctoPrint instances" checkbox with the current version? Does that make a difference?

dnblankedelman, 2019-06-26:
No worries. I'm currently on the road so I can't test this until this weekend. Will rev the plugin forward again and let you know if that makes a difference. By current version, do you mean HEAD on the 3.5 branch or the latest release?

fieldOfView, 2019-06-26:
Either should do, but HEAD on the 3.5 branch will get you some fixes too. If unchecking the checkbox does not work, you could try reverting only the zeroconf.py file to the 3.5.4 version.

dnblankedelman, 2019-06-26:
Ok, will try these things as soon as I'm back and will report in. Thanks!

mastahype, 2019-06-27:
Interesting. I'd upgraded to Cura 4.1 a few weeks ago by overwriting my 4.0 app. Nothing regarding octoprint worked and I'm almost certain there was no option for Connect Octoprint. So, I rolled back to 4.0 since I had some prints to do. Today, out of frustration over bugs in the Cura 4.0 UI I decided to do a side-by-side install and keep both app versions. Cura 4.1 loaded up fine and since 4.1 has presets for my printer, it found that again and this time I was able to press the button for Connect Octoprint and it all came up, including the web cam stream. 
However, after issuing a "Print to Octoprint", the webcam stream disappears altogether and doesn't come back on the Monitor tab at all. I do see updated extruder and bed temps as they heat up, and constantly updating Printing Time and Estimated time left, contrary to @dnblankedelman experience. This is using 3.5.5 from the marketplace and Cura 4.1.0.

fieldOfView, 2019-06-27:
@mastahype so all in all, you are experiencing an entirely different issue. See #116

mastahype, 2019-06-27:
> @mastahype so all in all, you are experiencing an entirely different issue. See #116

Correct. I mostly chimed in to say my printer telemetry was indeed being refreshed on 4.1. I didn't realize the octopi camera was a known issue just yet or that I had it (until printing started). I dropped in the updated package you posted in #116 and it seems to be working well. Thanks.

dnblankedelman, 2019-07-01:
Ok, reporting in with a definite measure of success. 

Here's what I did (restarting Cura in between each step):

1. went to the plugin dir for Cura 4.1.0 and this plugin. Performed a ```git pull``` to bring it up to date with the 3.5 branch
1. removed the existing printer under the plugin settings
1. added it back with  "Automatically discover local OctoPrint instances" unchecked. Also used "octopi.local" for the hostname as I previously had it before this issue cropped up. (one small note: I did see some interesting behavior with the settings persist, i.e. it didn't require me to add the API key back in even though the printer had been deleted and Cura restarted). 

From that point on, I was seeing the monitor tab be fully responsive (both could preheat and then observe the temperatures rise). I'm doing a test print now and the reported print times info is also staying in sync with the octoprint web interface. 

Given these steps, I don't know if the issue was the checkbox (perhaps it still has issues) or later changes made to the code to correct other things, but I'm in a fine working state. If you do want me to isolate further (i.e. try the latest code but with the checkbox on), I'm happy to do so. If not, feel free to close this issue.

Thanks for your support and the great plugin!

fieldOfView, 2019-07-01:
I think there is some issue on your specific computer, or your specific network configuration that not only prevents the automatic discovery with zeroconf/bonjour from working, but that makes the version of the zeroconf implementation included with the plugin actually block/cause timeouts even when it is not used. It is good to know the plugin works as expected when zeroconf is disabled. I think the checkbox is a good-enough workaround for now, so I am closing the issue.

The API keys are cached/remembered on a per-instance basis. In theory, you could have multiple 3d printer instances in Cura all connect to the same OctoPrint instance. That is why the key data is not cleared when a printer is removed.

dnblankedelman, 2019-07-01:
> some issue on your specific computer, or your specific network configuration that not only prevents the automatic discovery with zeroconf/bonjour from working, but that makes the version of the zeroconf implementation included with the plugin actually block/cause timeouts even when it is not used

I have to admit it is equally interesting to me that all I needed to do was to revert to a commit before the zeroconf.py file got manually downgraded and things started to work again even with zeroconf enabled. It's even weirder to me because (without looking at the plugin code), I can imagine the zeroconf stuff probably only gets called under select circumstances and is not a code path taken unless you are adding a printer. If I had to pull an idea out of my hat, I would wonder if the older code shims or monkey patches other python network calls on import in some way that breaks things. Perhaps that got removed in later versions? But this is all pure speculation.

I think both of us are thinking we can live with this work-around. If you change your mind, let me know and I'll break out the python debugger and wireshark and figure out exactly what is going on. 

Thanks again.

closed 2019-07-24 by dnblankedelman