fieldOfView/Cura-OctoPrintPlugin #116: Can not see OctoPi camera in Cura 4.1

bill-orange, 2019-05-31:
I am running OctoPi with a Pi camera hooked up to the camera port. The camera can be seen through my network without any issues. I am running Cura ver 4.1 on PC which is able to see the camera. I loaded the plug-in, set up the API string, rebooted, etc. I can monitor the printer without any issues. All the data I expect shows up except I do not see the camera. There do not seem to be any settings to confirm that Cura is looking at the right IP. 

Can I fix this?  Suggestions?

fieldOfView, 2019-06-01:
Are you sure you checked the checkbox to display the camera in the Connect to OctoPrint dialog?

Can you post your logs somewhere (dropbox, google drive, pastebin)? See Help -> Show configuration folder. The file you are looking for is named Cura.log

bill-orange, 2019-06-01:
Double and triple checked it!

Here is the log:

https://1drv.ms/u/s!AmXKqAwyCrbxiL5bUYI5DGG-gGM-UQ

pyrokidd89, 2019-06-03:
I have the same issue.

bill-orange, 2019-06-04:
@pyrokidd89 Probably be a good idea to post your log file too.

pyrokidd89, 2019-06-04:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3249960/cura.log)

bill-orange, 2019-06-04:
This is common to both your log file and mine.

2019-06-03 19:55:27,481 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///E:/Programs/Ultimaker Cura 4.1/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null

dragon076, 2019-06-04:
Have the same problem as well. I have the creawsome add-on install now but was still seeing the same problem without it. What i did notice is before i start the print i go and reconnect the printer and un-tick and re-tick the webcam box sometimes it works again. Log attached.

[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3250606/cura.log)

fieldOfView, 2019-06-04:
It would be helpful if you could let me know if the same problem occurs with Cura 4.0 and the current version of the plugin (3.5.5).

bill-orange, 2019-06-04:
I have Cura 4.0 on a spare machine.  The current plugin works fine on Cura 4.0.

fieldOfView, 2019-06-05:
> The current plugin works fine on Cura 4.0.

Then I am going to have to hunt in Cura what changed to break things.

tjones99, 2019-06-06:
I had the same issue and just did two things so at first I wasn't sure what fixed it, but I just did a test and now know... 

I just did an Octoprint upgrade to 1.3.11... turns out that did not fix the issue... but nice to be current... 

What fixed it was reconnecting to Octoprint in Cura. I went in to manage printers, and into the connect to Octoprint dialog. There was already my auto discovered network instance of Octorpint named octopi._octoprint._tcp.local that I was connected to (and successfully printing and managing no issues) but I added a new connection and put in the specific IP address of my Octoprint server and I literally copied the API key from the octopi._octoprint._tcp.local and pasted it into my new connection I was making and clicked Connect and my webcam instantly appeared. 

I then went back and connected back to octopi._octoprint._tcp.local and webcam view disappeared. Reconnected to my newly created connection, back... 

So I got it working... but do not know why one works and the other doesn't...

bill-orange, 2019-06-06:
Interesting. That should give @fieldOfView a hint as to what change in Cura 4.1 upset the apple cart. I might be with posting the log where you switched from one to the other.

tjones99, 2019-06-07:
Attaching two log files. I deleted the log file, launched Cura and went to monitor and saw webcam and exited and saved that small log. 
Then I relaunched Cura and set it to the failing Octoprint connection, exited, deleted that log then relaunched Cura and went to monitor (no webcam) and exited. Smallest possible log. 

My view of the log tells me little about what is failing..

[curaoctoprint.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3264041/curaoctoprint.zip)

m0oml, 2019-06-07:
I have the same issue in 4.1 works fine in 4 if you go manage printers and click the connect octoprint button it comes back but I have to do it every time

tjones99, 2019-06-07:
If you create a new connection using the address instead of the auto discovered instance (helps if you assign a static Ip using a shop reservation in your router) then no need to disconnect and connect.

bill-orange, 2019-06-07:
Not a good long term solution but that does appear to work.

rmaes4, 2019-06-19:
+1 having the same problem

anduchs, 2019-06-23:
I can confirm. Manually adding Octoprint via IP works, whilst using autodiscovered Octoprint does not work. Cura 4.1 with Creawesome here.

fieldOfView, 2019-06-26:
This should be fixed with https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/1c5d42a2c00b7bb58578ea5817d6b48e4aee3235. I will do some more testing and publish an updated version to the Marketplace.

This was one of those pesky issues that did not manifest when running Cura from source. I also did not have an OctoPrint instance with a camera for a while, which made debugging this harder as you can imagine. Thanks for reporting, all.

fieldOfView, 2019-06-27:
Please test this version:
http://files.fieldofview.com/cura/OctoPrintPlugin-v6.0.0-2019-06-27T11_31_22Z.curapackage
To install, drop the downloaded file into a running Cura application window and restart Cura.

elboletaire, 2019-06-27:
I tried it yesterday and I can confirm it now works properly with autodiscovery (I'm updating from source).

fieldOfView, 2019-06-27:
I have submitted a fixed version to the Marketplace, which should be available soon(ish).

commit 1c5d42a2 (2019-06-26): Fix showing camera

Fixes #116
closed 2019-06-27 by fieldOfView