fieldOfView/Cura-OctoPrintPlugin #118: Unable to connect plugin due to SSL Client Certificates

ilovecats516, 2019-07-22:
Hi, I've configured my Octoprint server to require a SSL Client Certificate for connecting to it. I used [these](https://community.octoprint.org/t/restrict-access-to-octopi-utilizing-ssl-client-certificates/3681) instructions to set it up. Is there a way to get this plugin to connect despite needing the certificate?

Thanks.

fieldOfView, 2019-07-22:
Have you installed the SSL certificate system-wide?

There's a "use https" checkbox hidden in the "Show reverse proxy options (advanced)" section. Did you try that?

ilovecats516, 2019-07-23:
That doesn’t work unfortunately.

fieldOfView, 2019-07-24:
Could you upload your cura.log somewhere so I can have a look?
See help -> show configuration folder

ilovecats516, 2019-07-25:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/3432551/cura.log)

fieldOfView, 2019-07-26:
According to those logs, you did not change the port. Port 80 is normally used for HTTP traffic; for HTTPS/SSL, port 443 is more common.

ilovecats516, 2019-07-26:
That didn’t work.

fieldOfView, 2019-07-30:
Can you post a new log?

fieldOfView, 2019-08-12:
I really need the logs with port set to 443, or I cannot help you.

jarfil, 2020-12-10:
The plugin only supports basic auth over HTTPS as of now, not HTTPS Client Certificate auth.

Would be nice if it did, though.

fieldOfView, 2020-12-10:
@jarfil, please open a new issue for that