fieldOfView/Cura-OctoPrintPlugin #120: Plugin not showing up after installing in Cura 4.2

mikebreeuwer, 2019-07-30:
Cura 4.2 is not showing (can't configure it, don't see it) the activated plugin after picking and installing it from the marketplace.

iyotee, 2019-07-30:
Pour suivit, I got exactly same problem here and i'm blocked as i'm usely printing remotly

CBAMav, 2019-07-30:
same problem here

iyotee, 2019-07-30:
Ahaha i just did it and was here to write my review also.

pseudex, 2019-07-30:
this is already updated in the version in branch 3.5. you use the install instruction from the readme.md and replace the marketplace version with this one

fieldOfView, 2019-07-31:
A version of the OctoPrint Connection plugin compatible with Cura 4.2 is awaiting review. In the mean time you can download it here:
http://files.fieldofview.com/cura/OctoPrintPlugin-v6.0.0-2019-07-18T12_32_22Z.curapackage

Download the file and drop it in a running Cura application.

probonopd, 2019-07-31:
Same problem here.

```
me@host:~$ cat  ./.local/share/cura/4.2/cura.log | grep Oct
2019-07-31 13:51:34,035 - INFO - [MainThread] Toolbox.src.Toolbox.startDownload [538]: Attempting to download & install package from https://api.ultimaker.com/cura-packages/v1/cura/v6.1.0/packages/OctoPrintPlugin/download.
2019-07-31 13:51:34,364 - DEBUG - [MainThread] UM.PackageManager.getPackageLicense [521]: Found potential license file '/files/plugins/OctoPrintPlugin/LICENSE'
2019-07-31 13:51:36,124 - INFO - [MainThread] UM.PackageManager.installPackage [355]: Package [OctoPrintPlugin] version [3.5.6] is scheduled to be installed.
2019-07-31 13:51:46,441 - DEBUG - [MainThread] OctoPrintPlugin.__matchVersion [53]: This version of the plugin is not compatible with this version of Cura. Please check for an update.
2019-07-31 13:51:46,441 - WARNING - [MainThread] OctoPrintPlugin.register [28]: Plugin not loaded because of a version mismatch
2019-07-31 13:51:46,445 - ERROR - [MainThread] UM.PluginRegistry.loadPlugin [392]: Plugin OctoPrintPlugin did not return any objects to register
2019-07-31 13:53:35,684 - DEBUG - [MainThread] OctoPrintPlugin.__matchVersion [53]: This version of the plugin is not compatible with this version of Cura. Please check for an update.
2019-07-31 13:53:35,684 - WARNING - [MainThread] OctoPrintPlugin.register [28]: Plugin not loaded because of a version mismatch
2019-07-31 13:53:35,688 - ERROR - [MainThread] UM.PluginRegistry.loadPlugin [392]: Plugin OctoPrintPlugin did not return any objects to register
2019-07-31 13:57:05,001 - DEBUG - [MainThread] OctoPrintPlugin.__matchVersion [53]: This version of the plugin is not compatible with this version of Cura. Please check for an update.
2019-07-31 13:57:05,002 - WARNING - [MainThread] OctoPrintPlugin.register [28]: Plugin not loaded because of a version mismatch
2019-07-31 13:57:05,005 - ERROR - [MainThread] UM.PluginRegistry.loadPlugin [392]: Plugin OctoPrintPlugin did not return any objects to register
```

probonopd, 2019-07-31:
I can confirm that https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/120#issuecomment-516718924 is a working workaround for me.

b4545, 2019-08-01:
> A version of the OctoPrint Connection plugin compatible with Cura 4.2 is awaiting review. In the mean time you can download it here:
> http://files.fieldofview.com/cura/OctoPrintPlugin-v6.0.0-2019-07-18T12_32_22Z.curapackage
> 
> Download the file and drop it in a running Cura application.

This link does not work for me

cortices, 2019-08-01:
> A version of the OctoPrint Connection plugin compatible with Cura 4.2 is awaiting review. In the mean time you can download it here:
> http://files.fieldofview.com/cura/OctoPrintPlugin-v6.0.0-2019-07-18T12_32_22Z.curapackage
> 
> Download the file and drop it in a running Cura application.

Cura gives an error "Cannot open files of the type [...].curapackage" When dragging the .curapackage file onto Cura 4.2.0 or using "Open With...". Is there a different way to install it? Or just have to wait for it to propagate to marketplace?

Platform is macOS 10.14.5.

fieldOfView, 2019-08-01:
> This link does not work for me

@b4545, what about it does not work? Please be more specific or I cannot help you.

fieldOfView, 2019-08-01:
> Is there a different way to install it?

You could try File -> Open inside Cura, but I don't know if that works. If it doesn't, you are going to have to wait for the Marketplace release. I don't know what's the holdup.

cortices, 2019-08-01:
> You could try File -> Open inside Cura

Tried that, no luck. Will wait for marketplace release.

b4545, 2019-08-01:
> > This link does not work for me
> 
> @b4545, what about it does not work? Please be more specific or I cannot help you.

The file would not download yesterday. Sorry I can't remember if it was timing out or if there was a specific exception. I checked again this morning and the file is available again. Maybe a load issue yesterday. I didn't need it anyway becasue editing `plugin.json `made the plugin work again for me.

fieldOfView, 2019-08-02:
See this post on what's going on:
https://community.ultimaker.com/topic/29068-psa-octoprint-connection-plugin-in-cura-42-and-421/

fieldOfView, 2019-08-02:
Closing, since the updated plugin is now available in the Marketplace.

closed 2019-08-02 by fieldOfView