fieldOfView/Cura-OctoPrintPlugin #122: Switching between printer in Monitor tab crash Cura

steevegldev, 2019-08-01:
Cura Version: 4.2.1
Plugin version: 3.5.6

I have 2 OctoPrint server configured. One by printer. In Cura, if I click on Monitor tab to check webcam, then clicking on the to switch printer dropdown and select my second print, Cura crash.

Steps:
1. Open Cura
1. Click on Monitor tab. Webcam appear.
1. Click on dropdown menu and select an other printer.
1. Cura close by itself.

cura.log
```
2019-08-01 15:24:28,182 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-08-01 15:24:28,197 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdateChecker
2019-08-01 15:24:28,210 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdater
2019-08-01 15:24:28,220 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzReader
2019-08-01 15:24:28,232 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzWriter
2019-08-01 15:24:28,244 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeProfileReader
2019-08-01 15:24:28,266 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
2019-08-01 15:24:28,279 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-08-01 15:24:28,294 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-08-01 15:24:28,307 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-08-01 15:24:28,319 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-08-01 15:24:28,333 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-08-01 15:24:28,345 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MachineSettingsAction
2019-08-01 15:24:28,357 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ModelChecker
2019-08-01 15:24:28,370 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MonitorStage
2019-08-01 15:24:28,385 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PerObjectSettingsTool
2019-08-01 15:24:28,399 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PostProcessingPlugin
2019-08-01 15:24:28,413 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PrepareStage
2019-08-01 15:24:28,425 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PreviewStage
2019-08-01 15:24:28,442 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RemovableDriveOutputDevice
2019-08-01 15:24:28,467 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-08-01 15:24:28,482 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SliceInfoPlugin
2019-08-01 15:24:28,496 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SolidView
2019-08-01 15:24:28,510 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-08-01 15:24:28,535 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [112]: No auth data to retrieve the access_token from
2019-08-01 15:24:28,540 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin Toolbox
2019-08-01 15:24:28,555 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-08-01 15:24:28,571 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-08-01 15:24:28,588 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-08-01 15:24:28,606 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-08-01 15:24:28,619 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-08-01 15:24:28,637 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-08-01 15:24:28,690 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPReader
2019-08-01 15:24:28,706 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPWriter
2019-08-01 15:24:28,722 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UltimakerMachineActions
2019-08-01 15:24:28,816 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UM3NetworkPrinting
2019-08-01 15:24:28,835 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-08-01 15:24:28,839 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-08-01 15:24:29,186 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin USBPrinting
2019-08-01 15:24:29,352 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade21to22
2019-08-01 15:24:29,443 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade22to24
2019-08-01 15:24:29,542 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade25to26
2019-08-01 15:24:29,626 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade26to27
2019-08-01 15:24:29,725 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade27to30
2019-08-01 15:24:29,803 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade30to31
2019-08-01 15:24:29,855 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade32to33
2019-08-01 15:24:29,873 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade33to34
2019-08-01 15:24:29,892 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade34to35
2019-08-01 15:24:29,914 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade35to40
2019-08-01 15:24:29,932 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade40to41
2019-08-01 15:24:29,951 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade41to42
2019-08-01 15:24:29,968 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimpleView
2019-08-01 15:24:29,992 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin X3DReader
2019-08-01 15:24:30,017 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin XmlMaterialProfile
2019-08-01 15:24:30,035 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin XRayView
2019-08-01 15:24:30,061 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OctoPrintPlugin
2019-08-01 15:24:30,080 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OrientationPlugin
[... 229 lines cut ...]
2019-08-01 15:24:43,170 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [anycubic_i3_mega_extruder_0] added.
2019-08-01 15:24:43,173 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [anycubic_i3_mega_extruder_0 #2] added.
2019-08-01 15:24:43,534 - DEBUG - [MainThread] UM3NetworkPrinting.src.UM3OutputDevicePlugin.checkCloudFlowIsPossible [511]: Checking if cloud connection is possible...
2019-08-01 15:24:43,535 - DEBUG - [MainThread] UM3NetworkPrinting.src.UM3OutputDevicePlugin.checkCloudFlowIsPossible [533]: Cloud Flow not possible: User not logged in!
2019-08-01 15:24:43,538 - DEBUG - [MainThread] cura.PrinterOutput.NetworkedPrinterOutputDevice._createNetworkManager [313]: Creating network manager
2019-08-01 15:24:43,562 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.2/resources/qml/PrinterOutput/OutputDeviceHeader.qml:39: TypeError: Cannot read property 'name' of null
2019-08-01 15:24:43,565 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.2/resources/qml/PrinterOutput/OutputDeviceHeader.qml:39: TypeError: Cannot read property 'name' of null
2019-08-01 15:24:43,567 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.connect [296]: Connection with instance mega with url http://192.168.10.200:5002/ started
2019-08-01 15:24:43,571 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.2/resources/qml/PrinterOutput/OutputDeviceHeader.qml:39: TypeError: Cannot read [... cut ...]
2019-08-01 15:24:44,193 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.2/resources/qml/PrinterOutput/OutputDeviceHeader.qml:39: TypeError: Cannot read property 'name' of null
2019-08-01 15:24:44,201 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [369]: file:///C:/Program Files/Ultimaker Cura 4.2/resources/qml/PrinterOutput/OutputDeviceHeader.qml:39: TypeError: Cannot read property 'name' of null
2019-08-01 15:24:44,205 - DEBUG - [MainThread] cura.Machines.Models.BuildPlateModel._update [33]: Updating BuildPlateModel.
```

fieldOfView, 2019-08-03:
Could you check if this is fixed with the latest version available from the Marketplace with Cura 4.2.1?

See https://github.com/Ultimaker/Cura/issues/6007#issuecomment-517901102

steevegldev, 2019-08-03:
Confirm, fixed in 3.5.8. Thanks!

closed 2019-08-03 by steevegldev