fieldOfView/Cura-OctoPrintPlugin #126: OctoPrint not showing on Cura 4.3.0

spikeeee, 2019-09-10:
Within the Print settings 'Connect to Octoprint' is missing from the panel in Cura 4.3.0.  uninstalling/Reinstalling both Cura and the plugin, including manual and market installation does not make it appear.

Downgrading to 4.2.1 solves the issue

fieldOfView, 2019-09-10:
As before with the 4.2 version:

Each version of the OctoPrint Connection plugin has an built-in check to see what version of Cura it is running on: It specifies a minimum and maximum version it is compatible with. I check the plugin with each released version of Cura before submitting a version that is specifically marked as compatible with that version of Cura. I started doing this more than a year ago because on multiple occasions the plugin got the rug pulled from underneath it by changes to Cura between versions; at more than one point, having the previous version of the plugin installed with the new version of Cura would have stopped Cura from starting, even leading to certain versions of the plugin having to be specifically blacklisted in Cura.

I normally try to update the plugin during the beta period for a new version of Cura, so that the updated version of the plugin is available through the Marketplace when the stable version is launched. In this case I submitted the updated version a week before the launch of the stable version. Unfortunately, the time between the launch of the beta and stable versions of Cura is a busy time for the Cura team, so approval of new plugins sometimes gets delayed. There was an additional issue with the "contributors portal" where new versions of plugins are submitted which has further delayed the release of the updated plugin.

fieldOfView, 2019-09-10:
Fixed here: 2e588fdc5d0064ea1411a10cbc359b6fe6559807

jj-uk, 2019-11-16:
I'm running cura 4.3 and Octoprint plugin version 3.5.10 and cannot see the network printer.

Is cura 4.3 not comparable with 3.5.10 ?

ericnewman, 2020-09-20:
Any update on this?

fieldOfView, 2020-09-21:
This issue was fixed over a year ago, and the fix is part of what is currently in the Marketplace. If you have problem with the plugin now, please open a new issue.