fieldOfView/Cura-OctoPrintPlugin #128: Print doesnt autostart

ByteSizedMarius, 2019-09-25:
I've noticed a behavior where the print does not autostart. I'm not sure what causes this, but it mostly works the first time you start a print. When a print is finished and you upload another one, octoprint itself often waits for user-input and doesn't seem to start. 

Because of this, I'm thinking, that its either octoprints "fault" because its in the "print finished" state and doesn't process the commands fast enough or its the plugins fault for whatever reason. 

I had the feeling that it works when I restart cura between prints, but I couldnt yet certify this suspicion. It would be great if you could add some kind of fallback that checks whether the print has started after like 10 seconds of uploading the file.

ByteSizedMarius, 2019-09-26:
Restarting cura and changing the file name doesnt matter, also reconneting the printer doesnt change anything. Really strange behaviour now, idk whats going on anymore.

fieldOfView, 2019-09-26:
> When a print is finished and you upload another one, octoprint itself often waits for user-input and doesn't seem to start.

Can you elaborate on this? As long as the status in OctoPrint is still "printing", the plugin cannot start a new print. Can you confirm that the OctoPrint interface says the print is finished?

ByteSizedMarius, 2019-09-26:
I will change my octoprint lang to english, try to recreate the issue and record my screen for you.

ByteSizedMarius, 2019-09-26:
https://streamable.com/a7h4v

at like 10 secs you can see that the last file was uploaded like 15mins ago, and it finished successfully. I then upload the same file again without slicing it and it doesnt start. I then re-slice it and upload it again - wont start either. On other instances I waited a few mins to see if it starts , but it doesnt. 

I forgot to show it, but the checkbox "automatically start print" in cura is definitely ticked.

fieldOfView, 2019-09-26:
I need the logs of your octoprint instance: https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

I currently don't have a functional octoprint instance I can test/reproduce this with.

ByteSizedMarius, 2019-09-26:
I only have the generel log enabled. It doesnt seem to be very informative. Please tell me which logging-levels you would like to have.

https://pastebin.com/pzWT2xYB

ByteSizedMarius, 2019-09-29:
It started working yesterday and has been working since, Maybe it was just a fluke? Ill close this until I can reproduce it again.

closed 2019-09-29 by ByteSizedMarius

fieldOfView, 2019-09-29:
Thanks for the update.

I'm still considering handling the start of the print separately from the upload; currently I tell the API to upload, select and print the gcode file in one call. In some cases it is going to be beneficial to first upload through the API, and then start the print. That would also give me some more debugging options.