fieldOfView/Cura-OctoPrintPlugin #129: Add button to queue up a job to Octoprint instead of just print.

vastator69, 2019-09-28:
I dont know if it is feasible or not, but it would be nice to be able to send a job to Octoprint for storage instead of immediately print.   I realize you get a queue job button if Octoprint is busy with a job but it would be nice to get that queue button as well as the print button.   This way you can queue up jobs for Octoprint to do later.

coofercat, 2019-10-01:
Related: https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/113 and https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/7

My preferred solution would be to pull down the "print with Octoprint" selector and see "save file" (as now) and "send to octoprint" (or similar wording).

I realise that there's a "print after sending" option in the settings, but I'm thinking of more ad-hoc type work - perhaps sending a series of prints to the printer in another room before going over to it to prime it for the first print.

TriggerMeister, 2020-04-19:
I hope we are talking about the same thing.  Something changed, because it was doning that for me, but then it stopped.  I have had these settings for a long time:
[image]

...and when I clicked "print with octoprint" and there was not already an active print job, it would put it right in the print dialog box (see screenshot below), so I just had to click "print".  Now it stopped doing it and just puts it in the print queue.  Anyway, a button that would do the same thing would be great as well.  Thanks.

[image]

MarcinOrlowski, 2020-11-19:
@fieldOfView this is very useful feature - is there any blocker that this FRE is in limbo for over a year? Any hope to have this added?

fieldOfView, 2020-11-19:
Lots to do, limited time. Also I don't like having conflicting options (what you set in the Connect to OctoPrint dialog vs the output device you choose). There's also a conflicting feature I've been meaning to support, which is having multiple OctoPrint instances associated to a single printer. Then you would get the following list:
```
Save to File
Save to Removable drive
Print with OctoPrint #1
Send to OctoPrint #1
Print with OctoPrint #2
Send to OctoPrint #2
Print with OctoPrint #3
Send to OctoPrint #3
```
I think that looks confusing.

Personally I would rather combine this with an (optional) dialog where you can change the path and filename before printing (#199).

But eventually it mostly comes down to having a finite amount of time. Also I don't like being one of those people, but if you really want a feature like this you might consider [sponsoring me](https://github.com/sponsors/fieldOfView/). No promises, but I am more inclined to listen to people who actively show their appreciation.

MarcinOrlowski, 2020-11-19:
> Also I don't like having conflicting options (what you set in the Connect to OctoPrint dialog vs the output device you choose).

Why? If that'd be optional, I'd not mind as I'd have feature I like. Others can disable that to keep current behavior. Acceptable trade-off.

> Then you would get the following list [...]

Well, I do have just one printer connected ATM :) But again, if that feature would be optional, you could have it off if needed. Or maybe you could select which printers you want so be included?

> Personally I would rather combine this with an (optional) dialog where you can change the path and filename before printing

Again, could this be optional please? Because I actually do not like that idea of having separate dialog for things I could simply select from drop-down list, esp. considering I just have I printer so that would not be crowded place in my case.

> really want a feature like this you might consider sponsoring me. No promises, 

I appreciate your work and OctoPrint plugin is a huge relief, but I'd like to suggest additionally allowing sponsorship per specific feature.

tablatronix, 2021-04-21:
oneplus for queue option, I often queue while already printing, NOT the same as the settings option, as I want to do both still.

fieldOfView, 2021-04-21:
> I often queue while already printing

When you are already printing, the plugin tells you you cannot start another print *and give you a Queue button*.

tablatronix, 2021-04-21:
Because its confusing also you have to wait for it to timeout.

fieldOfView, 2021-04-23:
Sorry, what timeout?

tablatronix, 2021-04-23:
If I click print during a print, It sits there for like 20-30 seconds, then errors, then says add to queue

fieldOfView, 2021-07-06:
>  It sits there for like 20-30 seconds, then errors, then says add to queue

Sorry, what error?
Please, don't keep me dragging details out of you.

tablatronix, 2021-07-08:
The `cannot start another print and give you a Queue button` There seems to be a significant delay or a timeout sometimes before it comes up...

fieldOfView, 2021-07-08:
Thanks for the additional info.

> There seems to be a significant delay or a timeout sometimes before it comes up.

I think the plugin only receives feedback from OctoPrint after the upload is completed. So if you have a fairly large gcode file, or a slow connection with the OctoPrint instance, then that could explain the delay. I had not considered that. I guess it makes sense then to have a more direct way to queue.

> I think there is another popup above it also with an error sometimes, 

Please let me know if you reproduce that.

tablatronix, 2021-07-08:
Ill play with it tonight, I have not been printing lately at all.

melyux, 2026-09-22:
Any luck with this? Seems like a simple but huge QOF change