fieldOfView/Cura-OctoPrintPlugin #133: 3.5.10 + Cura 4.4.0 BETA : Cannot deactivate auto start print job and webcam image

yet-another-average-joe, 2019-11-20:
Cura 4.4.0 BETA + OctoPrint Connection 3.5.10
Windows 10
Tevo Tornado/Marlin 2.0
(only one printer, but 8 were created depending on the nozzle)

Reproduction steps :

Install 4.4.0 BETA
Connect a printer (OctoPrint dialog)
Uncheck auto start print job and webcam (I always deactivate these options, and they got activated after installation of 4.4.0 BETA) ; modifying these options for one printer changes them for all printers at once. But nothing is saved when closing the Preferences dialog.

Actual results :

These options are forced "on" by 4.4.0 beta. They can be changed but they are not saved. Closing and reopening Preferences, auto start + webcam are always back.

Expected results :

The user's choice should be used and saved

randellhodges, 2019-11-23:
I see this too. I believe it is related to the errors I see in the log.

```
2019-11-22 20:36:33,844 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [58]: Could not get metadata of container  because it was not found.
2019-11-22 20:36:35,768 - WARNING - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.setContainerMetaDataEntry [291]: Could not set metadata of container  because it was not found.
```

fieldOfView, 2019-11-23:
Thanks for confirming the bug and the detective work.

fieldOfView, 2019-11-23:
This should be fixed with 954da41ec172189dd8edf69608ee1b59b7d828d9

Could you test by replacing `%APPDATA%\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\qml\DiscoverOctoPrintAction.qml` with this version: https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/954da41ec172189dd8edf69608ee1b59b7d828d9/qml/DiscoverOctoPrintAction.qml

slundell, 2019-11-23:
I had the same issue, and replacing the file fixed it for me.

fieldOfView, 2019-11-23:
Thanks. Unless another incompatibility pops up, I'll submit a fixed version to the marketplace.

bryanhunwardsen, 2019-12-12:
@fieldOfView Im on 4.4.0(GA) w/ Plugin 3.5.10 dated 2019-11-08 (with no update available in the marketplace). Is there an update on this fix being published/flowing to end users? Thanks

fieldOfView, 2019-12-16:
There is now an update in the Marketplace. 

I wanted to ensure compatibility with the upcoming OctoPrint 1.4 before pushing the update, but that seems to be working now so there was no longer a reason to "hold back" this fix.

closed 2019-12-16 by fieldOfView