fieldOfView/Cura-OctoPrintPlugin #135: Octoprint Error: 503 Bad Gateway

Deneteus, 2019-11-28:
A newer version of Octoprint is out (OctoPrint: 1.4.0rc1) and when I use Cura 4.4.0 Stable when I send a file it now shows the files being sent by Cura and then ends with Bad Gateway Error.

[image]

[image]

fieldOfView, 2019-11-28:
Thanks for reporting. I am seeing the same on my 1.4.0 RC1 installation. I'm not sure at this point if it is something the plugin is doing wrong or OctoPrint.

fieldOfView, 2019-11-29:
FYI, https://github.com/foosel/OctoPrint/issues/3358. Could you check your octoprint.log to see if there is a similar error in your logs?

I don't know if it is relevant yet, but what do you run OctoPrint on? If OctoPi, then what version of OctoPi?

fieldOfView, 2019-11-29:
See https://github.com/foosel/OctoPrint/issues/3349. This should be fixed with OctoPrint 1.4.0 RC2

closed 2019-11-29 by fieldOfView

nikon662, 2019-12-01:
For a Quick fix I ran this code and restarted octoprint
`wget -O /home/pi/oprint/lib/python2.7/site-packages/octoprint/server/util/tornado.py https://raw.githubusercontent.com/foosel/OctoPrint/staging/devel/src/octoprint/server/util/tornado.py`
This is the first time I've tried to help someone on github so if I've done something wrong please let me know. Thanks :)

fieldOfView, 2019-12-02:
The problem is also fixed here: ab40c033cb36115575380c234340b893df091aad

fieldOfView, 2019-12-02:
OctoPrint 1.4.0 RC2 is now available, which should fix this bug from that side.

Deneteus, 2019-12-04:
I tested it yesterday and was able to send files with no problems with the RC2 update.