fieldOfView/Cura-OctoPrintPlugin #136: Octoprint busy

tablatronix, 2019-11-29:
I get this alot and have to restart cura.

Let me know what you need to debug, thanks..

[image]

https://github.com/Ultimaker/Cura/issues/6733

fieldOfView, 2019-11-29:
Thanks for “transferring” the issue here.

First step is to check if the printer is indeed connected and not printing or in an error state. Can you “jog” the printhead through the web interface? And through Cura?

tablatronix, 2019-11-29:
Yeah printer was fine, afaik

fieldOfView, 2019-12-02:
Are you sure that the OctoPrint instance is not actually still in the "cancelling" state? If you eg cancel a print while heating up, the heating up needs to finish before the printer accepts new commands. Thus the printer is not actually ready to print another print.

tablatronix, 2019-12-02:
Nope, I wonder if this has to do with the preheat plugin, I started using that more often.

How can I debug, can I manually check the api socket connection

fieldOfView, 2019-12-02:
A cura.log is useful (Help -> Show configuration folder -> cura.log). Please upload it somewhere (eg pastebin.com) and share a link here; it is too large to paste into a comment here.

When it happens again, please open http://octopi/api/printer?apikey=yourapikey in a browser and post the response here.

tablatronix, 2019-12-03:
Will do thanks!

Jzooor, 2019-12-15:
This issue happens to me quite frequently if the computer I had sent the print job from goes into standby during a print. The print will complete, and Octoprint itself will be idle but if I wake up my computer and try to start a new print Cura will show the message above. The monitor page will also show a print still in progress. If I queue the job and then jump over to Octoprint right away, the job I sent will already be waiting to print and I just have to click start. The monitor page will just keep on showing the values that were last updated before the computer went to sleep. I've tried reconnecting to octoprint in the manage printers and the web cam image shows up, but the temp and progress values remain stuck. Only way to recover seems to be to restart Cura.

fieldOfView, 2019-12-16:
@tablatronix could it be that your computer coming out of sleep/standby is also involved in your case?

marcelrv, 2020-01-05:
I experience exactly the same. Will try to do digging in the logfiles to see if something stands out.

fieldOfView, 2020-01-06:
@marcelrv, @tablatronix and @Jzooor, what OS do you use, and what version of Cura?

marcelrv, 2020-01-06:
Cura 4.4 on Win10 in my case

tablatronix, 2020-01-06:
osx, previous version, have not upgraded to latest yet, 4.4.?

Jzooor, 2020-01-07:
I am on Win10. On Cura 4.4.1 now, but I have not run any long enough prints to have my PC go to sleep since updating from 4.3.

marcelrv, 2020-01-11:
Here I have the output. The first time it was fine, the 2nd print I get a busy

```
{
  "sd": {
    "ready": true
  }, 
  "state": {
    "flags": {
      "cancelling": false, 
      "closedOrError": false, 
      "error": false, 
      "finishing": false, 
      "operational": true, 
      "paused": false, 
      "pausing": false, 
      "printing": false, 
      "ready": true, 
      "resuming": false, 
      "sdReady": true
    }, 
    "text": "Operational"
  }, 
  "temperature": {
    "bed": {
      "actual": 25.45, 
      "offset": 0, 
      "target": 0.0
    }, 
    "tool0": {
      "actual": 18.28, 
      "offset": 0, 
      "target": 0.0
    }
  }
}

```

fieldOfView, 2020-01-25:
I need a Cura.log to know what is going on. See Help -> Show configuration folder. Please upload cura.log somewhere and post a link here (as the file may be sizeable).

fieldOfView, 2020-01-28:
I happened to come across this video which exhibits a similar issue:
https://youtu.be/IwUz0kfMrSA?t=712

crysxd, 2020-03-02:
I encounter the same issue regularly. It can be "fixed" by going to Preferences > Printer > Connect OctoPrint and then clicking "Disconnect" followed by "Connect". The following extract from the cura.log shows how I tried to start a print but got the message "OctoPrint busy" and then completed above steps. After this I could start the print without issues.

    2020-03-02 21:16:06,046 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
    2020-03-02 21:16:06,049 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
    2020-03-02 21:16:06,056 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:79:9: QML Label: Binding loop detected for property "width"
    2020-03-02 21:16:06,060 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [663]: Slicing took 4.4107348918914795 seconds
    2020-03-02 21:16:06,060 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [664]: Number of models per buildplate: {0: 6}
    2020-03-02 21:16:06,061 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [678]: See if there is more to slice...
    2020-03-02 21:16:06,064 - DEBUG - [Thread-3] CuraEngineBackend.ProcessSlicedLayersJob.run [70]: Processing new layer for build plate 0...
    2020-03-02 21:16:06,389 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [181]: Socket state changed to Closing
    2020-03-02 21:16:06,392 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [203]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
    2020-03-02 21:16:06,395 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [183]: Socket state changed to Closed
    2020-03-02 21:16:06,415 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [829]: Backend quit with return code 0. Resetting process and socket.
    2020-03-02 21:16:06,548 - DEBUG - [Thread-3] CuraEngineBackend.ProcessSlicedLayersJob.run [250]: Processing layers took 0.48334598541259766 seconds
    2020-03-02 21:16:06,559 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [859]: See if there is more to slice(2)...
    2020-03-02 21:16:56,159 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.requestWrite [374]: Tried starting a print, but current state is printing
    2020-03-02 21:16:56,180 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [b08b3d23e6124a039df13c740d272f74] started
    2020-03-02 21:16:56,256 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
    2020-03-02 21:16:56,830 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [b08b3d23e6124a039df13c740d272f74] finished.
    2020-03-02 21:16:56,834 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
    2020-03-02 21:16:56,834 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
    2020-03-02 21:16:58,243 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
    2020-03-02 21:17:01,707 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///Applications/Ultimaker Cura.app/Contents/Resources/qml/UM/Dialog.qml:86:13: QML Loader: Binding loop detected for property "active"
    2020-03-02 21:17:02,320 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [215]: Trying to access OctoPrint instance at http://192.168.1.100:80/ with the provided API key.
    2020-03-02 21:17:02,654 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFinished [418]: API key accepted by OctoPrint.
    2020-03-02 21:17:04,330 - DEBUG - [MainThread] cura.PrinterOutput.NetworkedPrinterOutputDevice._createNetworkManager [305]: Creating network manager
    2020-03-02 21:17:04,346 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [156]: Output Device Ender3 already added
    2020-03-02 21:17:04,347 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.connect [303]: Connection with instance Ender3 with url http://192.168.1.100:80/ started
    2020-03-02 21:17:04,352 - DEBUG - [MainThread] cura.PrinterOutput.NetworkedPrinterOutputDevice._createNetworkManager [305]: Creating network manager
    2020-03-02 21:17:04,354 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.connect [303]: Connection with instance Ender3 with url http://192.168.1.100:80/ started
    2020-03-02 21:17:04,414 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [156]: Output Device Ender3 already added
    2020-03-02 21:17:04,417 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [156]: Output Device Ender3 already added
[... cut ...]
    2020-03-02 21:17:07,212 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
    2020-03-02 21:17:08,706 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
    2020-03-02 21:17:09,531 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [772]: Resource created on OctoPrint instance: http://192.168.1.100/api/files/local/CE3_6677a13f-2efa-4d92-88b5-be24003c29e2.gcode

crysxd, 2020-03-02:
Ok, just happend again. Here is the output of /api/printer. I also spotted that the "Monitor" tab is stuck. The printer is done but there it shows the print progress 33%.

    {
      "sd": {
        "ready": false
      }, 
      "state": {
        "flags": {
          "cancelling": false, 
          "closedOrError": false, 
          "error": false, 
          "finishing": false, 
          "operational": true, 
          "paused": false, 
          "pausing": false, 
          "printing": false, 
          "ready": true, 
          "resuming": false, 
          "sdReady": false
        }, 
        "text": "Operational"
      }, 
      "temperature": {
        "bed": {
          "actual": 33.75, 
          "offset": 0, 
          "target": 0.0
        }, 
        "tool0": {
          "actual": 25.45, 
          "offset": 0, 
          "target": 0.0
        }
      }
    }

fieldOfView, 2020-03-02:
The issue seems to be that the plugin stops receiving updates from OctoPrint, or that those updates have stale information. Since OctoPrint is showing correct data in your browser, and since your logs don't show anything that would suggest the plugin is no longer receiving any updates, I'm going to have to make a version that does more logging to know what is going on.

crysxd, 2020-03-03:
Feel free to send me stuff to test, the issue seems very common on my machine although I'm still not 100% sure what's the teigger

tablatronix, 2020-03-03:
I thought I had a way to reproduce the other day but it started working.. lol I did notice I have more issues when I restart octoprint for plugin updates etc, if that helps. Of course cura stays open during etc..

crysxd, 2020-03-04:
@fieldOfView I added logs to `_update(self)` and `_onRequestFinished(self, reply: QNetworkReply)`. When the UI is stuck `_update(self)` is still called every 2s and the requests to `/job` and `/printer` are still executed with a 200 response.

This means there must be some issue when handling the response and updating the UI.

I'm pretty sure by now that this issue is caused by the laptop going into standby for some time (maybe more than 10 minutes?).

fieldOfView, 2020-03-27:
I am still none the wiser *how* this can happen (especially after your helpful debugging, @crysxd), but I do have a hunch. When it happens again, could you try disconnecting the printer via the OctoPrint web interface, waiting 15 seconds, and connecting again? Does that restore a sane status in Cura?

crysxd, 2020-03-27:
I will try! Thanks for your support and engagement 👍

fieldOfView, 2020-03-27:
It seems more and more issues prop up that may have the same underlying mechanism, so your feedback and debugging is much appreciated.

crysxd, 2020-04-12:
Believe it or not, this issue does not occur any longer for me (or at least it seems like). The only thing I can imagine causing this is that I migrated my Octoprint server from a native installation on a Raspberry to Docker running on the same Raspberry.

I'll keep my eyes open and will try your fix if it occurs again. Let me know if I can do something else (let me check the logs actually, maybe I see something).

fieldOfView, 2020-04-12:
Interesting... That could still also point to network stability as a cause.

crysxd, 2020-04-12:
I doubt that it's a network issue as the logs sho that the requests are still performed with valid HTTP responses

WhiskeyTang0F0xtr0t, 2020-07-03:
I know this is an older thread, but this has been happening to me for a while and wanted to drop a note here. On a whim I unchecked "Automatically discover local Octoprint instances" and it hasn't happened since.

fieldOfView, 2020-07-04:
@WhiskeyTang0F0xtr0t thanks for reporting that, because I still have no way to reproduce this issue. 

I take it you manually added your OctoPrint instance?

WhiskeyTang0F0xtr0t, 2020-07-04:
Yessir.

fieldOfView, 2020-07-04:
Is that because automatic discovery just never seemed to work for you, or do you have a configuration where bonjour cannot work, eg due to needing https or basic authentication?

WhiskeyTang0F0xtr0t, 2020-07-04:
Bonjour works fine and I'm using a .local hostname in the plugin config. I was troubleshooting and noticed the list kept flickering, so I figured I would disable it since it wasn't needed and don't want unnecessary polling.

SteveDC2, 2020-09-05:
Old thread but still happening on 4.5.0 for me

I am getting this issue too. I tried disconnecting and re-connecting with Octoprint within Cura... no dice.
I have 2 printers though and I had just switched from slicing for one printer, then set it printing and switched to the second printer and sliced then tried to "print with Octoprint" and now keep getting the error.

So, ensuring I had printer 2 selected I sliced then added to the que, then selected "open octoprint" and it opened my FIRST printer preview!!!

Before this I went to check the API key for printer 2 and it didn't match what the web view of printer 2 Octoprint displayed. I copied the key from printer 2 octoprint instance to the printer 2 Cura octoprint setup page and it STILL would not print (i.e. still gave printer busy).

Then I deleted the model, re-loaded, re-sliced and BAM... I could then print to octoprint on printer 2. Before this I had not changed the model. I only switched between printers to print the same model on both printers.

This feels like bad caching to me. For me something is not correctly switching between printers correctly. I will try to keep an eye on this and see if I can be more scientific with how I get it into the messed up state.

Side note.... my PC HAD been sleeping after a print I set going last night so could still be linked to sleep somehow.

BR,
Steve

Eibwen, 2020-10-15:
This occurred for me in the following manor:

Printer: Ender 3

1. Have octoprint and cura all in a ready state
2. Forgot to turn the primary power supply On on the printer (so it was being powered via the USB connection only)
3. Hit "Print with Octoprint" in Cura
4. The printer tries to start heating up, but that pulls the voltage down from the USB so it soft-crashes the USB connection at very least
5. Turn on the power supply, and via the printer's interface Abort Job
6. Via the OctoPrint web interface, it shows "Disconnected" and the print job aborted
7. Via OctoPrint web interface, re-establish the connection
8. Go back to Cura, it *seems* in a good state
9. Press "Print with Octoprint" and get this error

(I have not specifically tried to reproduce that, but it happened to me at least twice with that same pattern.  But I can reproduce or share logs if that would be helpful)

From that state:
* If I say yes to "add to queue", it shows up in the OctoPrint web interface, and I can just go to that and hit "Print" in that web interface and everything works
* If I go into Manage Printers, and disconnect then reconnect via your settings interface, it will then return to normal

So from that behavior, it seems like the Cura Plugin is anticipating OctoPrint is in a bad state, but not automatically trying to call the API calls to get it back into sync.  So my suggestion would be whenever that error shows up, your application could try to automatically disconnect and reconnect?  Or some other way to basically reset to make sure all the data is consistent?

I'm assuming your plugin is keeping some `state` which doesn't seem to get updated, even though OctoPrint itself had it's state corrected