fieldOfView/Cura-OctoPrintPlugin #137: Feature Idea: Sync Gcode Usage (line nb?) from Octoprint to Cura Preview view

Xqua, 2019-12-07:
Octoprint has this really nice Gcode viewer that syncs with the Gcode usage. 
Cura has a nice Tool Path preview in 3D. 

What would be the way to link both ? Does Cura has an API to the Preview module ? Does OctoPrint ? 
If so it could be great to sync both of them up to see your model being constructed in 3D in Cura !

fieldOfView, 2019-12-07:
This is actually surprisingly less straightforward than you might think.

OctoPrint can be printing something else than what you have loaded into Cura. After you start a print to OctoPrint from Cura, you can continue working in Cura; if you eg change the layer height and reslice at that point, you can imagine that the layer data does not match what OctoPrint is printing. You can even close and later restart Cura, and it will no longer have the layer data.

In theory, Cura could download the gcode from OctoPrint, parse it and display it. But have you ever loaded a gcode file back into Cura? It is a slow and cpu-intensive process. The normal "preview" after slicing does not actually show parsed Gcode, but it shows "layer data" that is produces alongside the gcode. This is also why the preview does not show what is done in start & end gcode snippets.

OctoPrint does not know what line is being executed by the printer, just what line it is sending. The printer has a buffer, so there is usually an up to a couple of seconds difference between what OctoPrint sends and what the printer prints.

Xqua, 2019-12-16:
I see, well it was worth mentionning ! 

I'll close this then as it does seem like an edge case and probably not that useful given that it would lock a CURA instance up and probably has too many ramifications ! 

Thanks for your detailed anwser !!

closed 2019-12-16 by Xqua