fieldOfView/Cura-OctoPrintPlugin #138: [BUG] Octoprint 1.4.0 RC3 - Printer is Offline

Deneteus, 2019-12-13:
Issue: Octoprint 1.4.0 RC3 broke the ability to send files from Cura. Worked in 1.4.0RC2.
It shows that Cura is connected in the printer dialog but the printer error says it's offline.

[image]

[image]

I left some feedback here: 
https://github.com/foosel/OctoPrint/issues/3389

fieldOfView, 2019-12-13:
> It shows that Cura is connected in the printer dialog but the printer error says it's offline.

To clarify: the Connect to OctoPrint dialog says that Cura is connected to OctoPrint. The error message is OctoPrint saying it is not connected to the printer. Please confirm in the OctoPrint web interface that the printer is "connected" and that you can eg jog the head using the controls on the control tab.

Deneteus, 2019-12-13:
> To clarify: the Connect to OctoPrint dialog says that Cura is connected to OctoPrint. The error message is OctoPrint saying it is not connected to the printer. Please confirm in the OctoPrint web [...]

I can indeed Jog the head of the printer and preheat.

[image]

fieldOfView, 2019-12-16:
When it happens again, please open http://octopi.local/api/printer?apikey=yourapikey in a browser and post the response here.

fieldOfView, 2020-01-14:
Has this ever happened again?

fieldOfView, 2020-01-25:
I need a Cura.log to know what is going on. See Help -> Show configuration folder. Please upload cura.log somewhere and post a link here (as the file may be sizeable).

fieldOfView, 2020-01-28:
I'm going to close this issue because of lack of activity, but I think it is the same underlying issue as #136 which I'll leave open for now. Cura seems to somehow loose its connection and stop updating the status of OctoPrint.

closed 2020-02-14 by fieldOfView