fieldOfView/Cura-OctoPrintPlugin #139: Cura won't start if Octoprint is not connected?

Jakobud, 2019-12-22:
Lately I've experienced some issues getting Cura to start up. It gets stuck on the Loading Plugins message on the load screen.

I started looking through the logs and found this:

```
2019-12-21 17:17:05,442 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [116]: No auth data to retrieve the access_token from
```

After thinking about this a bit I realized that this could be the Octoprint plugin having issues if my printer is turned off or if I haven't gone into the web interface to connect Octoprint to the printer.

Is this what this is from? If I delete the Octoprint plugin then Cura starts up just fine... It would be nice to know if this is the issue. Also, Octoprint being online should not be a requirement just to start up Cura.

fieldOfView, 2019-12-22:
The OctoPrint plugin does not use the cura.OAuth2 module, so the log you point out cannot be the reason for Cura not starting. Please post a link to the entire log.

I run Cura with the OctoPrint Connection plugin and the printer turned off more often than not. There must be something else going on.

fieldOfView, 2020-01-13:
Without further information (such as a complete log), I can't confirm the OctoPrint plugin is at fault nor fix the issue. Feel free to reopen this if you have a cura.log file for me to have a look at.

closed 2020-01-13 by fieldOfView

Jakobud, 2020-01-13:
That's fine it's not been clear what the problem is.