fieldOfView/Cura-OctoPrintPlugin #141: Power on TP-Link Smartplug

bergsm, 2019-12-29:
Similar to #79, This feature would allow users to power on their printer connected to a TP-Link smartplug via the OctoPrint Cura Plugin. 

Link to Octoprint TP-Link plugin: [https://plugins.octoprint.org/plugins/tplinksmartplug/](https://plugins.octoprint.org/plugins/tplinksmartplug/)

fieldOfView, 2019-12-30:
While I'm at it, might as well support the Tasmota plugin too: https://plugins.octoprint.org/plugins/tasmota/

If I understand it correctly, I would need to add a field for the IP address of the plug (and a plug index for the tasmota plugin).

fieldOfView, 2019-12-30:
>I would need to add a field for the IP address of the plug

It turns out I can get that information from OctoPrint so I can present the user with something friendlier than an IP address field.

fieldOfView, 2020-01-04:
Two more forks of the TP-Link smartplug plugins that should be supportable with relative ease:
https://plugins.octoprint.org/plugins/tuyasmartplug/
https://plugins.octoprint.org/plugins/orvibos20/

fieldOfView, 2020-01-11:
More candidates:
https://github.com/jneilliii/OctoPrint-WemoSwitch
https://github.com/da4id/OctoPrint-MyStromSwitch

Also, https://github.com/jneilliii/OctoPrint-TasmotaMQTT has a different API than https://github.com/jneilliii/OctoPrint-Tasmota :-/

fieldOfView, 2020-01-12:
I've been working on support for the following plugins for automatically turning the printer on when uploading a print: TP-Link Smartplug, Orvibo S20, Wemo Switch, Tuya Smartplug, Domoticz, MyStrom Switch, Tasmota and Tasmota MQTT (along with PSUControl, which was already supported). One of these plugins is detected, the Connect to OctoPrint dialog will have an option to "Automatically turn on the printer" using a configured plug. After sending the API command, the plugin will wait until the printer is connected to OctoPrint and continue uploading the file.

Here's a development snapshot of the plugin: [CuraPackage for Cura 4.4.x](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-12T14%3A49%3A28Z.curapackage?Expires=1894200572&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=KGx%2BV9VGQHbbS2U6%2B9J5bBSscSHg5KzQU5ZSCBjqkHrTvOH5BRnQLKKHgPJBYDWXPnae8m%2BcVkLanroqoD%2Bgo2OzQBU4l8RZe88Qyjf1G6b8%2FcKria8nxXjEukOnYq3JcPw2CZspuvx3aiJLnXcwb4cJn3W4CIR6%2FZCUK%2BtdsOBQSMLvZxmHiggoNdRDOX4f2zIk%2FRi8c0HLFnLfIzcPLRCkPMTUttvchLK3Gr9CZSKdcgTaQwY1f4lx1N2czaS0GrblBjUSR3ZuzoNKhdGiIYhecq64B7JaUvdm8c4JISpvesExSCpRbRQ0BPxJ2Nr8Y9L7OL%2FPpNGLIwIUIpQGPg%3D%3D). Download the file, and drop it into the running Cura application window (not into a folder!). Then restart Cura.

thelastWallE, 2020-01-12:
> I've been working on support for the following plugins for automatically turning the printer on when uploading a print: TP-Link Smartplug, Orvibo S20, Wemo Switch, Tuya Smartplug, Domoticz, MyStrom [...]

this snapshot was not working because of the cmakelists.txt. It is not installing the files for the new plugfunction.
Will try now to install manual from github with zip

fieldOfView, 2020-01-12:
If you use the curapackage linked above, the CMakeList.txt file is not used. So that cannot be the problem.

Please provide a log file if the snapshot is not working.

thelastWallE, 2020-01-12:
steps taken:
uninstalling plugin
cura restart -> plugin was gone in the pluginfolder
droping the package in cura
cura restart -> plugin was installed but it seems it is installing the 3.5.11 version.
i see no new function for the switch. also reading 3.5.11 on the connect window.

In the marketplace it reads 3.5.11 is it getting it from there?

[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4050838/cura.log)

fieldOfView, 2020-01-12:
Sorry, the Contribute portal ate my update. Here's a new link to the [CuraPackage for 4.4](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-12T16%3A38%3A23Z.curapackage?Expires=1894207106&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=kqcwMCSQBS%2Ff1PSaTJQz5zYCgyj%2F8veJwCosqpyerzOmknBWHUixLclfLmmhF5LDtT25XuQRngR4GN5849ZIs8pytCtjfSrV1wE46Sjc6eK4u1NTr87FJMsgRcAFnWjDWz2uvr6NAheUIn01LZKEIbQkhLaQ9QNYJ1r%2FVma7BZ5M3gA9wDXJfqU3ZNPriUTgTA5hNyAhZHgfnM3aUmQEjGPy29vu0CkKdCMJyTBNieLl33lav1m55LqVTTP0e%2BcjQfF6t9DRZdhMvsB%2BpRMamVdCGi6WlW8DHfHpLpkaZ8%2FesQ03IqLJfVIZMukCLYNX1XvXoIy6vHzdbyvZjgutdw%3D%3D)

thelastWallE, 2020-01-12:
i have now import errors.
cura log:

`2020-01-12 17:53:56,809 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin LinearAdvanceSettingPlugin
2020-01-12 17:53:56,809 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Import error loading module OctoPrintPlugin
2020-01-12 17:53:56,825 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2020-01-12 17:53:56,825 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 540, in _findPlugin
2020-01-12 17:53:56,825 - ERROR - [MainThread] UM.Logger.logException [88]:   File "imp.py", line 245, in load_module
2020-01-12 17:53:56,825 - ERROR - [MainThread] UM.Logger.logException [88]:   File "imp.py", line 217, in load_package
2020-01-12 17:53:56,825 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 693, in _load
2020-01-12 17:53:56,841 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2020-01-12 17:53:56,841 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap_external>", line 697, in exec_module
2020-01-12 17:53:56,841 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2020-01-12 17:53:56,841 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\__init__.py", line 6, in <module>
2020-01-12 17:53:56,841 - ERROR - [MainThread] UM.Logger.logException [88]:     from . import OctoPrintOutputDevicePlugin
2020-01-12 17:53:56,841 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevicePlugin.py", line 26, in <module>
2020-01-12 17:53:56,856 - ERROR - [MainThread] UM.Logger.logException [88]:     spec.loader.exec_module(zeroconf)
2020-01-12 17:53:56,856 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 41, in <module>
2020-01-12 17:53:56,856 - ERROR - [MainThread] UM.Logger.logException [88]:     import ifaddr
2020-01-12 17:53:56,856 - ERROR - [MainThread] UM.Logger.logException [88]: ImportError: No module named 'ifaddr'
2020-01-12 17:53:56,856 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [615]: Could not find plugin OctoPrintPlugin
2020-01-12 17:53:56,873 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Import error loading module OctoPrintPlugin
2020-01-12 17:53:56,873 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2020-01-12 17:53:56,873 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 540, in _findPlugin
2020-01-12 17:53:56,873 - ERROR - [MainThread] UM.Logger.logException [88]:   File "imp.py", line 245, in load_module
2020-01-12 17:53:56,873 - ERROR - [MainThread] UM.Logger.logException [88]:   File "imp.py", line 217, in load_package
2020-01-12 17:53:56,889 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 693, in _load
2020-01-12 17:53:56,889 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2020-01-12 17:53:56,889 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap_external>", line 697, in exec_module
2020-01-12 17:53:56,889 - ERROR - [MainThread] UM.Logger.logException [88]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2020-01-12 17:53:56,889 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\__init__.py", line 6, in <module>
2020-01-12 17:53:56,904 - ERROR - [MainThread] UM.Logger.logException [88]:     from . import OctoPrintOutputDevicePlugin
2020-01-12 17:53:56,904 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevicePlugin.py", line 26, in <module>
2020-01-12 17:53:56,909 - ERROR - [MainThread] UM.Logger.logException [88]:     spec.loader.exec_module(zeroconf)
2020-01-12 17:53:56,909 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 41, in <module>
2020-01-12 17:53:56,909 - ERROR - [MainThread] UM.Logger.logException [88]:     import ifaddr
2020-01-12 17:53:56,909 - ERROR - [MainThread] UM.Logger.logException [88]: ImportError: No module named 'ifaddr'
2020-01-12 17:53:56,925 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [406]: Loaded plugin PauseBackendPlugin`

thelastWallE, 2020-01-12:
there are import errors now on plugin loading:

[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4050861/cura.log)

fieldOfView, 2020-01-12:
Thanks, at least we're getting somewhere now ;-)

fieldOfView, 2020-01-12:
Here's a [new version](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-12T17%3A24%3A07Z.curapackage?Expires=1894209850&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=FhjHOiYDFQaJunNoksBb%2Bhh0ioIPhCOkztq25Wib%2FHoZ0nmNM5LPNdq%2FunMOyZsDQ6SrjVjoqWjHdNYIr4sPgEqvKb6XbPq96%2F81mjOd8Xeh0seJ0TNMYq6dKHv5l%2BUvtbD%2BXeaH8ClmUYPC4Xqma4WGYXMF%2B0wdwSR%2FXyeiKmG%2B%2BYe%2BQO2Obry2MEbUqXeh4OYhyVmuCUe3iTPqWlaSfx0JudlWQe8%2ByE0WT4fhHWq2m%2B7aBcvirbtuQ6mPVEytodv2sFdnecqH5dxAqyamZZSCmlLqlBnmrJV%2FQkk0RthnJtRvdHHpBzHus6UoP8OHbAhOni5KjEO%2BcSSspUVi4w%3D%3D) that includes the module that could not be imported.

thelastWallE, 2020-01-12:
Seems like it doesn't like the "," comma.
`[MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevicePlugin.py", line 39
2020-01-12 18:30:27,982 - ERROR - [MainThread] UM.Logger.logException [88]:     except FileNotFoundError, ImportError:
2020-01-12 18:30:27,984 - ERROR - [MainThread] UM.Logger.logException [88]:                             ^
2020-01-12 18:30:27,984 - ERROR - [MainThread] UM.Logger.logException [88]: SyntaxError: invalid syntax`

https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/b93be74a68cbcb801e18ef6489ec2d66b147823b/OctoPrintOutputDevicePlugin.py#L39
`except FileNotFoundError, ImportError:
    # fall back to the system-installed version, or what comes with Cura
    from zeroconf import Zeroconf, ServiceBrowser, ServiceStateChange, ServiceInfo`

fieldOfView, 2020-01-12:
Perhaps I should not be doing this on a sunday evening afterall... It should have been `    except (FileNotFoundError, ImportError):`

Here's a [final version](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-12T17%3A46%3A56Z.curapackage?Expires=1894211220&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=eMlvCX6rucHOIHliaPxqD%2Fb7O7bb61u2gUUDmGsd%2FbrbdhVRa%2F%2BckW7vaFFJZd79hvKlFX%2BUL4ZQWQv42rt7YG5o7xQRFPIJ4vIS2bYFQZo78c4xkjvm9ifS8rJ3r77qThBeTqwABJbaAzRA2vzGkv7wOYc7IhQNWZbgZHrmwgT2guSh3F227XiE6yJ2Ip8MDz3oa4bijpA3GGvxGepO5chuL2Kmews8gv1ztVJrxh2csknc3rki%2Fn30dtuL4KKjWTevJ2FmKeUmjtufDcodYyJ2MqWfKrEl8A3JdCz3cOsJfYxxYpBLtHjSyzGSlmnD8q03%2BEgwhp95J%2FF3iFX1kA%3D%3D) for today.

thelastWallE, 2020-01-12:
nice. on the same time. 💯

thelastWallE, 2020-01-12:
Yeah should be enough for today for me, too. My wife gets already angry. :-D

Just for next time:
```
2020-01-12 19:08:37,678 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Unable to write to file retracttest: Can't convert 'collections.OrderedDict' object to str implicitly
2020-01-12 19:08:37,678 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2020-01-12 19:08:37,678 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 135, in _writeToDevice
2020-01-12 19:08:37,678 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevice.py", line 398, in requestWrite
2020-01-12 19:08:37,678 - ERROR - [MainThread] UM.Logger.logException [88]:     set_state_command = self._power_plugins_manager.getSetStateCommand(True, power_plug_id)
2020-01-12 19:08:37,694 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Users\Sven\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintPowerPlugins.py", line 75, in getSetStateCommand
2020-01-12 19:08:37,694 - ERROR - [MainThread] UM.Logger.logException [88]:     end_point = "plugin/" + plugin_id
2020-01-12 19:08:37,694 - ERROR - [MainThread] UM.Logger.logException [88]: TypeError: Can't convert 'collections.OrderedDict' object to str implicitly
```
Clicking on "Print with Octoprint" does nothing.

fieldOfView, 2020-01-13:
A new [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/UW44dINkkhO5BDyONGJIr9M1_GsYLNbEAeSAjNp3YJgG/OctoPrintPlugin-v7.0.0-2020-01-13T13%3A55%3A46Z.curapackage?Expires=1894283749&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=hB5Bggs%2F4vEtcN%2Bht8f8YNXtC88NOI0AKwajMwVh1RtNoi5gEOgwa1Gh58GqomBpVMCl5oeOKP%2B5yI%2FXKb1OD8kfHiWY2wEo%2Fnhh%2FVPYWxHLBV3VCdp8Vgff%2BciiRbsdXJ7dEU3VlNCg0EiRu6PcQFvqkHYdNDZsxkP5GU8bhVBS%2BnwP6F8ITqIJ6pBh2sNMXgtxpbDaaIwGuz8i3I7ZHY6oXD87eczmmyDenC3lsBWHpnuTqL4NoNSPsjzO6ynZNMg4hGk2R2kKYK1KwY%2F0qgPPgVUFVF2HSyKigfvdtRNpDO2AUb9VupetUAG8hEW%2Bh2OgSkUj0eqOiSeKYDk0uw%3D%3D) should fix the remaining issues with the TPLink Smartplug support. I have not tested the other plug types, because I don't have them.

thelastWallE, 2020-01-14:
First try and it works. Really nice! This is so good.

fieldOfView, 2020-01-17:
This seems to be working now.

fieldOfView, 2020-01-19:
Some work remains to be done.

@jneiliii did some testing here: https://community.octoprint.org/t/cura-octoprint-connection-plugin/5348/147

TPLink works, the others not so much (yet).  As said in that topic, I don't have most of the smart plugs, so I rely on your feedback. I also sometimes make assumptions that might not be correct.

The interesting logic happens here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/6504037dc320ac38e2ea5458ea789cf9376cbbed/OctoPrintPowerPlugins.py#L12

In this file, the "plugins" part of the OctoPrint settings dict is parsed to detect smart plug plugins. The data for each available plug is checked per specific plugin.

The `getSetStateCommand` further down the file gets the API endpoint and json data to POST to that endpoint. There might very well be data missing from the json.

fieldOfView, 2020-01-19:
The long and short of it is that these are the arguments that are sent to the "turnOn" endpoint for each plugin. The values are taken from the `arrSmartplugs` array:
```
        if plugin_id in ["tplinksmartplug", "orvibos20", "wemoswitch"]:
            arguments = ["ip"]
        elif plugin_id == "domoticz":
            arguments = ["ip", "idx"]
        elif plugin_id == "tasmota_mqtt":
            arguments = ["topic", "relayN"]
        elif plugin_id == "tasmota":
            arguments = ["ip", "idx", "username", "password"]
        elif plugin_id == "tuyasmartplug":
            arguments = ["label"]
```
@jneilliii I take it that for the (non mqtt) Tasmota plugin, I also need the `backlog_delay` argument, and for the Domoticz plugin I also need a `user` and `password` arguments.

I was looking at the arguments listed in the `get_api_commands` dict before, but I see now that there can be additional arguments.

Wondro, 2020-02-18:
I tested this with the 0.8.9 version of the Tasmota plugin and got it to power the printer on. I had to remove the `backlog_delay` or it would crash when you tried to add the plug.

fieldOfView, 2020-02-18:
@Wondro thanks for testing. Could you elaborate a little?

> I had to remove the `backlog_delay`
What did you remove where?

> or it would crash
What would crash? Cura? OctoPrint? Tasmota?
If Cura, could you send me a link to your Cura.log?

Wondro, 2020-02-18:
I removed it from
`("tasmota", "Tasmota", ["idx", "username", "password"]), # ip, idx, username, password` 

and

`        elif plugin_id == "tasmota":
            # ip, idx, username, password
            arguments = ["ip", "idx", "username", "password"]`

Cura would crash with this error when it was in there
`Traceback (most recent call last):
  File "C:\Users\Wondr\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\DiscoverOctoPrintAction.py", line 524, in _onRequestFinished
    self._power_plugins_manager.parsePluginData(json_data["plugins"])
  File "C:\Users\Wondr\AppData\Roaming\cura\4.4\plugins\OctoPrintPlugin\OctoPrintPowerPlugins.py", line 48, in parsePluginData
    plug[key] = plug_data[key]
KeyError: 'backlog_delay'`

fieldOfView, 2020-02-18:
Thanks, I'll get that fixed

jneilliii, 2020-02-19:
Thanks @fieldOfView, the web based Tasmota plugin release he is referring to is nearing final stages, and it re-worked a lot of server-side stuff including removing the backlog_delay, as well as username/password in the api call, so it should function more like the tplink one once officially released.

fieldOfView, 2020-02-20:
Fixed in https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/0e57e7b9e00f08f9cf4a7ce60950c9634ea06bd1. Should be available in the Cura Marketplace later today.

fieldOfView, 2020-02-20:
This feature is now available in the Marketplace:
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/OctoPrintPlugin

closed 2020-02-20 by fieldOfView

bergsm, 2020-03-14:
Finally got around to using the marketplace version. Works beautifully! 👍