fieldOfView/Cura-OctoPrintPlugin #142: Timeout with Octoprint behind nginx reverse proxy via https

Bruceforce, 2020-01-04:
Hi,

I got a timeout while checking the API key against an octoprint instance which is running behind an nginx reverse proxy via https. In the cura plugin i checked the reverse proxy and the `Use HTTPS` checkboxes. My client trusts the certificate of the proxy.

Unfortunately the `cura.log` isn't quite helpful since it only provides the following:
```
2020-01-04 11:25:15,407 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.connect [303]: Connection with instance Octoprint with url https://192.168.0.75:9443/octoprint/ started
2020-01-04 11:25:15,916 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [215]: Trying to access OctoPrint instance at https://192.168.0.75:9443/octoprint/ with the provided API key.
```

here's the reverse proxy config:
```
    location /octoprint/ {
      proxy_pass http://192.168.0.65:5000/;
      proxy_set_header Host $http_host;
      proxy_set_header Upgrade $http_upgrade;
      proxy_set_header Connection "upgrade";
      proxy_set_header X-Real-IP $remote_addr;
      proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_set_header X-Scheme $scheme;
      proxy_set_header X-Script-Name /octoprint;
      proxy_http_version 1.1;

      client_max_body_size 0;
    }
```

Note that opening `https://192.168.0.75:9443/octoprint/api/printer?apikey=MY_API_KEY` in a browser works just fine.

OctoPrint plugin version: 3.5.11
Cura version: 4.4.1
OctoPrint version: 1.3.12

fieldOfView, 2020-01-04:
> I got a timeout

Did you get an actual message telling you about a timeout, or did it just take forever?

Bruceforce, 2020-01-04:
It did just take forever - i.e. nothing happened. Sorry for making this not clear.

fieldOfView, 2020-01-06:
The HTTPS handling does not get much testing, so it might very well be buggy. I'll try to look in to it.

The next release of the plugin will do a better job of handling timeouts, telling the user about it and adding more information about errors to the logs.

fieldOfView, 2020-01-13:
Are you using a self-signed certificate?

Bruceforce, 2020-01-13:
Yes I'm using a self signed certificate on the nginx reverse Proxy.

closed 2020-01-13 by fieldOfView

fieldOfView, 2020-01-13:
I've added what I think is a fix here: 2830fa7

commit 2830fa7b (2020-01-13): Fix connecting via SSL with self-signed certificate

Fixes #142