fieldOfView/Cura-OctoPrintPlugin #143: [Cura 4.4.1 / plugin ver. 3.5.11] Plugin not correctly installing/updating?

thelastWallE, 2020-01-12:
I have missing files if i let cura update installing the plugin. (OctoPrintPowerPlugins.py; NetworkReplyTimeout.py)
i tested with manual installing but i had problems with discovering the octoprint instance on the network.
i wanted to have the nice new tplinkswitch option.
Or is this not in the current release? It is installing 3.5.11

fieldOfView, 2020-01-12:
Are you installing from the Marketplace, or through some other means?

thelastWallE, 2020-01-12:
nvm i just saw that you have a developer snapshot on @ #141

fieldOfView, 2020-01-12:
I keep forgetting that there are people that use installation methods that directly pull from the github repository and thus require a correct CMakeList.txt. I fixed that.

closed 2020-01-12 by fieldOfView