fieldOfView/Cura-OctoPrintPlugin #145: Support inverted controls

fieldOfView, 2020-01-28:
OctoPrint has options to switch the jog control directions. The OctoPrint Connection plugin does not currently follow these preferences.

Requested here: https://community.ultimaker.com/topic/30991-octoprint-plugin-ender-5-plus-z-axis-reversed-in-monitor/?tab=comments#comment-255850

The configuration is retreived from `/api/printerprofiles`, which can list multiple profile but only one is `current`:
```
{
  "profiles": {
    "_default": {
      "axes": {
        "e": {
          "inverted": false,
          "speed": 300
        },
        "x": {
          "inverted": false,
          "speed": 6000
        },
        "y": {
          "inverted": false,
          "speed": 6000
        },
        "z": {
          "inverted": false,
          "speed": 200
        }
      },
      "color": "default",
      "current": true,
```

rapzak, 2020-02-13:
I like that :) Seems to have same issue, i can invert the Z axis in Octoprint, and runs fine there, but in cura it still is opposite...

DodgeDeBoulet, 2020-02-21:
> 
> 
> I like that :) Seems to have same issue, i can invert the Z axis in Octoprint, and runs fine there, but in cura it still is opposite... ( but why does octoprint not make it globat, that it is inverted in the printer setting??)

I believe it's not global in that there are printers that move the hot-end in the Z axis, and printers that move the bed in the Z axis. My Ender 5 plus does the latter, and the Z-axis control on the printer itself moves the bed in the selected direction. My preference would be for the buttons in OctoPrint (and the Cura plug-in) to behave similarly.

If I had another printer where the hot-end moved along the Z axis, its front-panel Z axis control would most likely move the hot-end in the selected direction. In that case I would want OctoPrint and the Cura plug-in to mimic _that_ behavior.

Make sense?

fieldOfView, 2020-02-21:
> I believe it's not global in that 

That's why the behavior is configurable in OctoPrint, and the issue at hand here is that the Cura plugin is ignoring that configuration. The next version of the OctoPrint Connection plugin will adhere to the configuration in OctoPrint, so the jog controls in both it and Cura behave the same way.

fieldOfView, 2020-05-26:
It took me a little longer to implement this, but it should work after this commit:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/146e5fa6a159204f6408c82b503159b623ae337d

DodgeDeBoulet, 2020-05-26:
LOL, so now I'm running a firmware (Tiny Machines 6.2, based on Marlin 2.0) that **doesn't** invert the Z-axis on the front panel ... (sigh)

The problem for me is that I think the inversion is in the screen code and isn't configured by any of the #defines in the Marlin headers 🤔

But thank you VERY much, regardless 😁