fieldOfView/Cura-OctoPrintPlugin #146: CURA autostart after uploading ufp file didn't work

OllisGit, 2020-02-03:
Hi @fieldOfView,  @jneilliii,

I am not sure where is the right place to raise the issue, so I mentioned both of you and also created in both githubs a ticket. Maybe it needs a "dynamic duo-solution". 
[image]

Problem:
When the UFP Plugin is installed in octoprint the "automatically start print job" feature didn't work. After disabling the plugin the autostart is working again.
 
Analyse:
I looked into the cura-plugin, because my first assumption was that the implementation calls the upload api with the print parameter=true. So only one call for upload, select file  and start print. I thought the unzipping is not done and so the file-selection couldn't work.
But the cura-plugin send a second print call....and looking into the sourcecode I found there is some octoprint-plugin specific magic.
https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/3.5/OctoPrintOutputDevice.py#L837
```
if "PrintTimeGenius" in plugin_data:
    self._wait_for_analysis = True
```
Hey, maybe I can use this as a workaround, just installing "PrintTimeGenius" and Cura is waiting a few seconds before start printing....but that didn't help.

Maybe you have an idea, why the autoprint feature is not working.

Thanks in advance,
Olli

fieldOfView, 2020-02-03:
I blame my plugin, so you posted in the right place.

The problem is that OctoPrint should start printing a different file (the extracted .ufp.gcode) than the file that was just uploaded (the .ufp file). This has been an issue since I first started supporting the @jneilliii's plugin, and is still an issue with the version of my plugin that is currently in the Marketplace.

I thought I had fixed this in the not-yet-released version that is now in the 3.5 branch, but apparently it needs some more finesse. It would be helpful if you could post your cura.log (see Help -> show configuration folder).

OllisGit, 2020-02-03:
Hi @fieldOfView ,

WOW, that was fast! Problem solved!
I just installed the current version from the 3.5-branch and now it is working!

Thanks a lot
Olli

fieldOfView, 2020-02-06:
Good to know the upcoming release does fix the issue.

closed 2020-02-06 by fieldOfView

Goodfeat, 2020-02-17:
Hello, I was looking for a solution to one problem and ended up on your discussion thread.

after installing the Ultimaker Format Package plugin on an octoprint, printing automatically stopped starting.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage

I have the latest version of Octoprint installed
[image]

it seems this has been fixed earlier, but now it does not work again

Cura 4.5 + octoprint 1.3.12

fieldOfView, 2020-02-17:
> I don’t understand why the version indicated in the marketplace version is the same as in the branch,

It isn't. The current version on the 3.5 branch of this repository is 3.5.12-DEV. The version in the Marketplace is 3.5.11. Version 3.5.12 of the plugin is awaiting approval for the Marketplace, but should be available from there "soon" (I expect somewhere this week).

sugar012, 2020-03-12:
Hello all!
With Cura 4.5 and Octo 1.4 seems to be broken again. I got the message "wait for octoprint to complete gcode analisys" and it goes forever. of course i can press the "print now" button .. in fact it's a minor annoyance but sometimes i find myself minutes later with the printer not moving and wondering why :)

fieldOfView, 2020-03-12:
>  I got the message "wait for octoprint to complete gcode analisys" and it goes forever

This dialog is shown if you have Print Time Genius installed. PTG does not work if you start the print before the gcode analysis is complete. Therefore the plugin probes OctoPrint to see if gcode analysis for the file has been completed before starting the print. It could be that there's a bug on my end, or perhaps the gcode analysis is never started after extracting the gcode from the ufp file. I'll try to find out.

jneilliii, 2020-03-12:
The ufp plugin tries to force the analysis once the file is extracted, but not sure how that works with PTG or not.

fieldOfView, 2020-03-13:
I have opened a new issue for this:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/158