fieldOfView/Cura-OctoPrintPlugin #147: Zeroconf bundled by octoprint plugin requires netifaces

nallath, 2020-02-05:
So I was just testing the octoprint plugin on the marketplace, but it seems to require netifaces, which fails to be imported.

Now, this seems to be caused by your modified / bundled version of Zeroconf. Is there still a reason for it to be there? We recently had some speed improvements merged into the main-line of zeroconf.

fieldOfView, 2020-02-05:
What version of Cura did you use? Cura 4.4 comes with netifaces. It also comes with a version of zeroconf that is buggy (eg it won't work/stop discovery if there are hosts with dots in their names, which never happens for Ultimaker machines, but does happen for OctoPrint instances). That's why I started to include a patched version of zeroconf with the plugin. This always worked because the plugin-included version has the same dependecies as the cura-included version. Will Cura 4.5 no longer include netifaces?

The prerelease version (current 3.5 branch) of the OctoPrint plugin includes the latest release of Zeroconf (with your patches). I could now stop distributing OctoPrint with the plugin, if it weren't for the great architectural decision of the Marketplace that forces me to make a single version that is compatible with all Cura versions including those with older buggy versions of Zeroconf. Because of this I also now have to include a version of ifaddr, because older versions of Cura need to be compatible with this newer version of Zeroconf. You can see why I am loving this Marketplace architecture.

nallath, 2020-02-05:
Hmm, then I need to poke the build system because the nightly 4.5 build wasn't working. 

I'm also not a fan of the Marketplace architecture, but it is what it is now.

nallath, 2020-02-05:
This can be closed, we switched to a different version of Zeroconf (which didn't rely on netifaces anymore). We will ensure that it's in the build system again.

closed 2020-02-05 by nallath