fieldOfView/Cura-OctoPrintPlugin #149: Positive x-axis jog button missing

BenBergman, 2020-02-17:
This may be an issue with my configuration somehow, but I don't have a positive x-axis jog button in Cura when connected over OctoPrint Connection. Is this a config issue? Plugin bug? Cura bug?

[image]

I'm running Cura 4.4.1, OctoPrint Connection 3.5.11, and OctoPrint 1.3.12.

fieldOfView, 2020-02-17:
That's something I have not seen before...

What OS do you use, and what screen resolution & DPI?

BenBergman, 2020-02-17:
I'm on Arch Linux. I have two displays, one is 22" 1080p and the other is 27" 1440p. The button is missing regardless of the screen Cura is on.

BenBergman, 2020-02-17:
To clarify on the DPI, both displays are set to 100% scaling, so no "retina scaling" or whatever on hiDPI displays.

fieldOfView, 2020-02-17:
Could you test if the problem also happens with the AppImage?

fieldOfView, 2020-02-17:
I think this may be an upstream bug in Cura, which does not manifest with the Qt version that is used in official distributions. But since I caused it, I'll see about fixing it.

BenBergman, 2020-02-17:
Huh, yeah, right X-axis arrow button shows up fine with the AppImage version of Cura.

fieldOfView, 2020-02-17:
This should be fixed by https://github.com/Ultimaker/Cura/pull/7126

closed 2020-02-24 by fieldOfView

Ghostkeeper, 2020-02-25:
Arch Linux is also using Qt 5.13? Or are you on 5.14 already?

BenBergman, 2020-02-25:
I've got Qt 5.14 on my system already. I'm not sure what version was running when I first noticed the issue as I've run system updates a few times since, but I am still seeing it now with Qt 5.14 and Cura 4.4.1.

fieldOfView, 2020-04-03:
The above-mentioned fix is now available in a public beta of Cura 4.6