fieldOfView/Cura-OctoPrintPlugin #150: use hostname instead of resolved IP address

fiveNinePlusR, 2020-02-22:
It would be great to have the plugin connect to cura with the hostname instead of the ip address since ip addresses are inherently volatile and it's a pain to have to login every time the ip address changes.

fieldOfView, 2020-02-23:
Please elaborate.

When adding an instance manually, you can in fact already enter a hostname instead of an IP address. 

When using the automatic discovery, the plugin always gets the current IP address from the instance and it identifies the instance to use by its zeroconf/bonjour/avahi name, so you don't ever have to change the IP address.

fieldOfView, 2020-03-02:
Closing due to inactivity.

closed 2020-03-02 by fieldOfView

fiveNinePlusR, 2020-03-02:
Sorry, I missed this earlier. 

The ip address is used to open the webpage instead of the hostname. so instead of what i fill out in the field as "octopi.local" it will send me to http://192.168.0.x

i am not using the automatic discovery.

fieldOfView, 2020-03-02:
If you don't use the automatic discovery, you added the instance address manually. In the address field, you can enter `octopi.local` instead of `192.168.0.x`. If you do, the plugin will use that to open the webpage.

fiveNinePlusR, 2020-03-02:
ah when i added it manually i didn't connect the hostname version. all is well thanks!

fieldOfView, 2020-03-02:
😺