fieldOfView/Cura-OctoPrintPlugin #151: Printing from Cura to OctoPrint is Causing Possible Stepper Issues

barnowan, 2020-02-24:
My Ender 3 prints fine from SD and from OctoPrint directly.  If I try to send a print to OctoPrint through this Cura plugin, the movement of all axes seems very rapid compared to normal operation, and the extruder tries to move out of the print area before the print even begins causing stepper motor issues until I kill power to the printer.

Please let me know what diagnostic information you need, and thanks in advance!

fieldOfView, 2020-02-24:
Unless this does not happen when you save the gcode and then upload it through the web interface, you should probably post this over at the OctoPrint issue queue: https://github.com/foosel/octoprint/issues.

barnowan, 2020-02-24:
This does not happen when I save the gcode and upload through the OctoPrint web interface.  It only happens when trying to print through OctoPrint through this Cura plugin.

A couple additional notes/observations:

- I do not use the stock Ender 3 control board.  I have replaced it with the SKR Mini E3 V1.2 (It runs Marlin like the stock one did).
- If I print through OctoPrint, it prints fine.  If I print through this plugin, the axes steps/mm seems off.  After I abort the failure print through Cura, if I try to print directly through OctoPrint again, it still has the stepper issues.  The 3D printer needs to be rebooted for it to function properly again.

After a full reboot of everything (OctoPrint, Ender 3, PC), everything seems to be functioning properly now.  If it happens again, I'll try to get the M92 values afterwards to see if those were being modified for some reason.

fieldOfView, 2020-02-24:
> This does not happen when I save the gcode and upload through the OctoPrint web interface.

Ok, that's weird. Can you try the following:
* prepare a smallish model and slice it
* save the gcode to file
* also send the gcode file to OctoPrint using the plugin
* (cancel the print)
* download the file gcode from OctoPrint
* zip up both gcode files and post them somewhere so I can compare the two

barnowan, 2020-02-25:
If the issue returns, I'll be sure to go through those steps and report back.  For now, it seems to be working fine after rebooting all components again.  I appreciate you checking in.

fieldOfView, 2020-03-27:
I'm closing this issue for now, since I am unable to reproduce it. If it does happen again, please keep commenting here (try to do the steps I mentioned above) and I'll reopen the issue.

closed 2020-03-27 by fieldOfView