fieldOfView/Cura-OctoPrintPlugin #152: Cura 4.4.1 - hanging on startup

jj-uk, 2020-02-24:
There seems to be an issue with the octoprint plugin in cura 4.4.1 where it causes cura to hang "updating configuration" during cura startup.

I installed the plugin, used it for a few hours. Restarted the PC and cura is hanging on startup.

Removing the plugin from the plugins folder and reinstalling fixes the issue- until the next day.

fieldOfView, 2020-02-24:
Please post a link to your cura.log file so I can see what is going on. It can be found via Help -> Show configuration folder. Upload the cura.log file somewhere and post a link here so I can download it.

fieldOfView, 2020-02-24:
What version of the plugin do you use, and how did you install it?

jj-uk, 2020-02-24:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4247300/cura.log)

I'm installing optoprint pl

ugin from the marketplace. I'm using the current version: 3.5.12

fieldOfView, 2020-02-25:
Thanks for the log. It took me a bit to find the actual hang-ups in the logs because they are so... sudden? As in: there is no indication in the logs that anything happens after the OctoPrint plugin (succesfully) loads.

Two things to try:
* Install this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlugin-v7.0.0-2020-02-23T13%3A58%3A58Z.curapackage?Expires=1897979116&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=Hp%2FO12D2srf4ltNvlw5iETr2lN5OnKO4doQlLNW1POeuE94Td4wVs6aed1C2kjlZUxdju2z3zRX5OezkLxGYjPNVyG8ZnBfXea6bEBpQ55z9cmKZuqBINgLto8O46T9bg2%2B0OgxeX4lqenvpez99Dm26YsIhAPnjT89H9p07sRblY4JaRCqHDZ%2FduOuxD6nKUFsriWJQShVG9LJ6hUBnnouqlDklS0PWyMLuS0yubVxeCVU2dQqHwfbJDt0sLahmHT8TsTC0CeLhomepourDLyVWwMcHWPuFwMBEvuEN%2B%2BQmvaCZVslVp4h%2B5UOFKa0OxG57%2B%2BHMcv16qJc1py5OXQ%3D%3D) by downloading the file and dragging it into the Cura window build volume.
* Disable the Bonjour/zeroconf discovery module. Go to the Connect to OctoPrint dialog, and untick the "Automatically discover local OctoPrint instances" checkbox at the bottom of the list.

jj-uk, 2020-02-25:
OK, i'll try this later, and report back in a few days if it works.

fieldOfView, 2020-03-02:
Have you been able to have a look?

jj-uk, 2020-03-02:
I've opened Cura a few times but its not locking up, i've not done anything since reporting this.

fieldOfView, 2020-03-08:
It is unclear from your comment if you disabled the "Automatically discover local OctoPrint instances" option and/or if you are now using the development snapshot.

jj-uk, 2020-03-09:
I've not done anything as i'm currently unable to reproduce it again. I'll be printing again tonight so I'll see what happens then.

jj-uk, 2020-03-15:
Today, Cura locked up at startup again usuing the release version of the plugin, so I've now installed the dev branch and unticked autodiscovery (which works fine).

How do I "Disable the Bonjour/zeroconf discovery module" ?

fieldOfView, 2020-03-15:
> I've now installed the dev branch and unticked autodiscovery (which works fine).

Sorry, what works fine? The dev branch, or autodiscovery? Or the dev branch with the autodiscovery unticked? I just want to be sure.

> How do I "Disable the Bonjour/zeroconf discovery module" ?

By unticking the autodiscovery, like you did.

fieldOfView, 2020-03-15:
Is your computer in a large network with many (apple?) computers?

jj-uk, 2020-03-15:
Ah ok, the autodiscovery has always worked fine in the past.

I'm printing now. I'll be doing a bit of printing over the next week or two and will report back if cura locks up on startup over that period.
It seems to be random when it locks up. Once it has locked up, it never starts again without needing to delete the plugin manually.

jj-uk, 2020-03-15:
> Is your computer in a large network with many (apple?) computers?

No, there are no apple devices on my network.

fieldOfView, 2020-03-27:
Something else to try is to keep the autodiscovery for the OctoPrint plugin enabled, but go into the Installed tab of the Marketplace and disable the "Ultimaker Network Connection" plugin.

jj-uk, 2020-04-01:
I've not had a problem since doing as instructed:

- Install this development snapshot by downloading the file and dragging it into the Cura window build volume.
- 
- Disable the Bonjour/zeroconf discovery module. Go to the Connect to OctoPrint dialog, and untick the "Automatically discover local OctoPrint instances" checkbox at the bottom of the list.

jj-uk, 2020-04-17:
Hi- the problem is still happening. Seems quite random.

fieldOfView, 2020-04-17:
This is with the "Automatically discover local OctoPrint instances" checkbox unticked?

jj-uk, 2020-04-18:
Yes, with the checkbox unticked.   I've installed cura 4.5 now. The changelog suggests there are changes to the way it loads plugins. I'll keep this post updated.

fieldOfView, 2020-04-18:
I have contributed those changes; it is mostly a speedup, but since your issue happens so randomly, that speedup could be what stops the issue from happening if it is a race condition. I'm interested to hear if it fixes anything for you.

fieldOfView, 2020-04-28:
Is this still happening with Cura 4.5 or 4.6?

jj-uk, 2020-04-28:
It has not happened again. I'll update this thread if it does.

When I installed the development snapshot by "downloading the file and dragging it into the Cura window build volume", does that mean I am now always on this snapshot or will it auto-update to the latest release?

fieldOfView, 2020-04-28:
It will not "auto-update", but the marketplace will show a notification when an official update is available, like it does when you have a normal version installed.

jj-uk, 2020-05-28:
This no longer seems to be an issue since cura v4.5.

closed 2020-05-28 by fieldOfView