fieldOfView/Cura-OctoPrintPlugin #154: Uploaded gcode not automatically selected on upload???

azalanono, 2020-02-28:
Hello.
I have configured that a printjob will be sent to the printer but not automatically be started. This always worked in that way, that the gcode file was also automatically selected when the printer was not printing.
A very handsome feature which seems to be removed.

So I have to send now the job to octoprint and then search it in list of gcode files every time.

Can't you get the old behavior back?
If there are reasons for not automatically select the uploaded file, please add a checkbox so I can enable/disable "automatic gcode selection when printer idle".

fieldOfView, 2020-02-28:
This behavior has not changed in any (recent) version of the plugin. If you want your files to be selected after upload, you might try the Autoselect plugin for OctoPrint: https://plugins.octoprint.org/plugins/autoselect/

You can also sort the file list by most recent upload. That way you don't have to search.

azalanono, 2020-02-28:
Hello.

In this case - that nothing was changed at cura in this point - I am really surprised why it 
does not work as before. 

     *  I click "Print with Octoprint" as I do all the time.
     *  The file is transmitted to octoprint as before.
     *  I open Firefox and see the old/previous printjob selected now. Here I saw the recently 
uploaded job in the past during last years, but this changed during last 2-3 updates in Cura.

I cannot safely update octoprint.    Gina added so much unnecessary crap to recent 
versions of octoprint that it is very close to exceed the resources of my "Raspberry Pi B", 
where it is running on. It already needs literally 5 minutes to get up and running.  So I 
disabled most of this crap and also further updates years ago. Octoprint has definitely not 
changed.

Am Freitag, 28. Februar 2020, 09:14:35 CET schrieb Aldo Hoeben:
> This behavior has not changed in any (recent) version of the plugin. If you
> want your files to be selected after upload, you might try the Autoselect
> plugin for OctoPrint: [...]

-- 
 
mit freundlichen Grüßen
 
  Ing. Rainer Hantsch
    (Firmeninhaber)
 
--
-------------------------------------------------
 ING. RAINER HANTSCH - Hardware & Software
 Khunngasse 21/20
 A-1030 Wien
 *** Austria ***
 
 Tel.: +43-1-7988538-0     Fax   : +43-1-7988538-18
 Mobil: +43-664-9194382    Skype : rainerhantsch
 UID-Nr: ATU11134002

fieldOfView, 2020-02-28:
For now, try this option:
[sort by upload date](https://user-images.githubusercontent.com/15846310/36679675-16bb7234-1b14-11e8-8cf6-8f3490c3c8f7.png)

I'll look in to the option of selecting gcode files.

> I cannot safely update octoprint.  

Then fix your setup. Better power supply, better cabling.

> Gina added so much unnecessary crap to recent versions of octoprint

I advice you to use different words and to be more appreciative of the time and energy put in to the software that you use without any cost to you.

azalanono, 2020-02-28:
Am Freitag, 28. Februar 2020, 09:41:04 CET schrieb Aldo Hoeben:
> For now, try this option:
> [sort by upload date](https://user-images.githubusercontent.com/
15846310/36679675-16bb7234-
> 1b14-11e8-8cf6-8f3490c3c8f7.png)
> 
> I'll look in to the option of selecting gcode files.
Thank you. But I think this must be related to Cura, I haven't changed anything on octoprint 
since years.

> > I cannot safely update octoprint.
> Then fix your setup. Better power supply, better cabling.
You misunderstood.  My setup is fine, good power supplies and cabling. But I do not risk to 
change anything on octoprint again and (possibly) running into new troubles with updates 
or plugins. I need this printer almost 24/7 and my new one is not finished yet (but it will). 
The "Raspberry Pi B", as I mentioned, is close to its limits since I upgraded to octoprint 
1.3.10; before it booted up 10 times faster.

> > Gina added so much unnecessary crap to recent versions of octoprint
> I advice you to use different words and to be more appreciative of the time
> and energy put in to the software that you use without any cost to you.
Sorry for my harsh words, but the final result is still the same: <60 seconds boot time on 
versions before 1.3.10 (IIRC till 1.2.9?), now 5+ MINUTES. :(
Adding more and more "bells and whistles" and requiring therefore repeatedly more 
powerful hardware is not always the right way. 
My friend ran into similar troubles and finally switched to "Repetier-Server Pro" to get this 
to an end. He says it was worth this little money: Same hardware, lots faster. But I prefer the 
octoprint UI, and as my printer is permanently powered up, I can accept this horrible boot 
time.
About "... without any cost ...": Software is free , but as it forces me to upgrade my hardware, 
it is not.

fieldOfView, 2020-02-28:
I've thought about this some more. Please read this issue, where someone is requesting the direct opposite of your request: https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/113

Your case van be solved by installing a plugin on the OctoPrint side of things (even though you don't want to). There is no such workaround for the other case.

I am reluctant to add a checkbox for this behavior because
* I think it makes the interface more complex than it needs to be
* I think it makes the plugin code unnecessarily complex

I am now of the opinion that the current behavior is more in line with the default behavior of OctoPrint. I'll leave it like this for a while and see what others think. I'll leave this issue open for civilized discussion about the matter.

fieldOfView, 2020-05-01:
I have added an option in the Connect to OctoPrint dialog to select uploaded files after upload. Can you please help test this functionality by installing this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlugin-v7.0.0-2020-05-01T10%3A28%3A25Z.curapackage?Expires=1903688909&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=Yb4MHsY%2FV6Tr66ot0QQ85JND4awqHtpPvuUoYPYCtNcQj8Eln4hEPrsSOTEmyHdhqcQ1RLB4kS55xDv1abcOJ99bMazsJ0vLZ4zJVhZsCG4BQdVX%2Fbfar%2Fn3JBUENXD2y48kINqVi1AsHcemLNW4OVbqs1ncyxjyXd8vIh6yg82KXhTpicTtZM3xlCKdvUEFa63n3Vx2beFy9KH1N0xRnQ%2FT8wYcbEcr6%2F7qZIr%2FLSJgjDEpXjOvV21Nun7VGlpf2z%2BDaNuxiudV4BKPfiOsUkLyBbTtgWFjo0GUXjgA%2FSmZSyEIWHFWvV4G8%2FVdRw%2BXx9fb0ljKIumkTGOdFhYEww%3D%3D)? Download the file and drop it onto the Cura buildplate as if you are opening a 3d model.

azalanono, 2020-05-01:
Hello.

I would like to test it for you, but how do I get this snapshot installed?

I am currently using Cura 4.5 as appimage on Linux. 64 bit.

Am Freitag, 1. Mai 2020, 12:30:38 CEST schrieb Aldo Hoeben:
> I have added an option in the Connect to OctoPrint dialog to select uploaded
> files after upload. Can you please help test this functionality by
> installing this [development
> snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/package
> s/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlug
> in-v7.0.0-2020-05-01T10%3A28%3A25Z.curapackage?
Expires=1903688909&GoogleAcce
> ssId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=Yb4MH
> sY
%2FV6Tr66ot0QQ85JND4awqHtpPvuUoYPYCtNcQj8Eln4hEPrsSOTEmyHdhqcQ1RLB4kS55x
Dv
> 1abcOJ99bMazsJ0vLZ4zJVhZsCG4BQdVX%2Fbfar
%2Fn3JBUENXD2y48kINqVi1AsHcemLNW4OVb
> qs1ncyxjyXd8vIh6yg82KXhTpicTtZM3xlCKdvUEFa63n3Vx2beFy9KH1N0xRnQ
%2FT8wYcbEcr6
> %2F7qZIr%2FLSJgjDEpXjOvV21Nun7VGlpf2z
%2BDaNuxiudV4BKPfiOsUkLyBbTtgWFjo0GUXjg
> A%2FSmZSyEIWHFWvV4G8%2FVdRw%2BXx9fb0ljKIumkTGOdFhYEww%3D%3D)? 
Download the
> file and drop it onto the Cura buildplate as if you are opening a 3d model.

-- 
 
mit freundlichen Grüßen
 
  Ing. Rainer Hantsch
    (Firmeninhaber)
 
--
-------------------------------------------------
 ING. RAINER HANTSCH - Hardware & Software
 Khunngasse 21/20
 A-1030 Wien
 *** Austria ***
 
 Tel.: +43-1-7988538-0     Fax   : +43-1-7988538-18
 Mobil: +43-664-9194382    Skype : rainerhantsch
 UID-Nr: ATU11134002

fieldOfView, 2020-05-01:
Like I said: Download the file and drop it onto the Cura buildplate as if you are opening a 3d model.

azalanono, 2020-05-01:
Sorry.  

I stopped reading at "download ...".    

I can confirm, it works perfectly.

Am Freitag, 1. Mai 2020, 13:36:11 CEST schrieb Aldo Hoeben:
> Like I said: Download the file and drop it onto the Cura buildplate as if
> you are opening a 3d model.

-- 
 
mit freundlichen Grüßen
 
  Ing. Rainer Hantsch
    (Firmeninhaber)
 
--
-------------------------------------------------
 ING. RAINER HANTSCH - Hardware & Software
 Khunngasse 21/20
 A-1030 Wien
 *** Austria ***
 
 Tel.: +43-1-7988538-0     Fax   : +43-1-7988538-18
 Mobil: +43-664-9194382    Skype : rainerhantsch
 UID-Nr: ATU11134002

fieldOfView, 2020-06-04:
Version 3.5.13 is now available in the Marketplace.

closed 2020-06-04 by fieldOfView