fieldOfView/Cura-OctoPrintPlugin #155: "OctoPrint error" dialog box

probonopd, 2020-03-07:
In recent versions of the plugin I am seeing this dialog box that was not there previously. 

[image]

What in the response causes this error box to appear?

Reference:
https://github.com/probonopd/WirelessPrinting/issues/142#issuecomment-596099081

fieldOfView, 2020-03-07:
The "OctoPrint error" dialog is shown when OctoPrint returns a HTTP status code > 400, and it is not an obvious/known case that can be handled more elegantly.

When OctoPrint returns a status code > 400, the body is normally text. It looks like your implementation returns a valid json response with an error HTTP status code. This sounds weird, because you either encounter an error, or you can produce a valid response. Not both.

probonopd, 2020-03-07:
Wireshark shows 200 though:

[image]

fieldOfView, 2020-03-07:
Please post a cura.log somewhere.

probonopd, 2020-03-07:
[cura.log.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4302157/cura.log.zip)

fieldOfView, 2020-03-07:
The log is not showing the specific error you posted above, but there is a different error that looks like a 404 error. Unfortunately there's a bug in the logging of the released version of the plugin which makes it unclear what endpoint the plugin was accessing when it encountered that error.

Could you apply this locally, and test again?
https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/337f2578f0dbf3d32457e3446898d74fad8e8687

probonopd, 2020-03-07:
[cura.log.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4302184/cura.log.zip)

probonopd, 2020-03-07:
`/api/login` is not implemented in WirelessPrinting since it had not been necessary so far.

fieldOfView, 2020-03-07:
A blank page would do, but would log a warning in cura.log. `{}` would be handled without an error.

This does not solve the mystery of the original error though. Can you still reproduce that?

fieldOfView, 2020-03-07:
My bad, the plugin is looking specifically for a 201 status code, as that is what OctoPrint returns when succesfuly uploading a file:
https://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder

I'll make the error checking a bit less strict for the next version, because a 200 response is not an error. It might take a couple of weeks before that gets submitted and reviewed by UM though, so it is probably quicker to fix it on your end by returning a 201 status instead of a 200 status code like the original specification suggests.

probonopd, 2020-03-07:
Thank you __very__ much for your help @fieldOfView. Will make the suggested changes on my end, too.

closed 2020-03-07 by probonopd