fieldOfView/Cura-OctoPrintPlugin #156: Print does not start automatically anymore

probonopd, 2020-03-07:
In recent versions of the plugin the print does not start automatically when used with https://github.com/probonopd/WirelessPrinting. https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/3cef0a955ae7ccfa5c07d20d9d147c530cc9d6ec (version 3.5.11) is known good. Somehow I suspect that https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/ab40c033cb36115575380c234340b893df091aad might have to do with it.

Reference:
https://github.com/probonopd/WirelessPrinting/issues/142#issuecomment-596099081

fieldOfView, 2020-03-07:
The more likely reason is https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/b197012c0702d37b9977164924d64151a495ed6c

The plugin now issues a separate command to the API to start the print after upload is complete, to support cases where the file that needs to be printed is a different file than the file that is uploaded (eg a file extracted from an archive) or cases where starting to print the file need to be delayed (eg until after gcode analysis is done).

See https://docs.octoprint.org/en/master/api/files.html#issue-a-file-command

closed 2020-03-07 by fieldOfView

probonopd, 2020-03-07:
Thank you very much for your explanation @fieldOfView. In WirelessPrinting, for now we will just start printing anything after uploading. This is sufficient for now. (Unless some volunteer implements the additional command to the API to start the print after upload is complete.)