fieldOfView/Cura-OctoPrintPlugin #157: Zeroconf does not find printer in 3.5.13-DEV (works in 3.5.12)

probonopd, 2020-03-07:
[image]

[image]

fieldOfView, 2020-03-08:
Could you use some actual words to describe your issue? Add some logs, that sort of thing.

probonopd, 2020-03-08:
Sorry, I assumed it was obvious from the screenshots. So here is what I did:
* Run Cura 4.5 AppImage
* Close Cura
* git clone Cura-OctoPrintPlugin to the plugins directory as described in the readme so that I get the current `master`

When doing this, the result is that Cura does not show any printers recognized by Zeroconf in the left-hand side of the window. So I did this:

* Close Cura
* git checkout <the hash that corresponds to 3.5.12>
* Restart Cura

And voilà, the printer recognized by Zeroconf appeared the left-hand side of the window.

fieldOfView, 2020-03-08:
LOGS, I asked specifically about logs. Unless you provide them, I will close this issue.

probonopd, 2020-03-08:
Give me some minutes. Learned something today: _Always_ attach the logs.

probonopd, 2020-03-08:
Cannot reproduce anymore. Yesterday I was running on Xubuntu 18.04, today I am running on Ubuntu 19.04. Will investigate more, please don't spend any time on this until I can safely reproduce.

Thanks for the great work you are doing all the time! :+1:

fieldOfView, 2020-03-08:
Sorry for snapping at you. The thing is, I write this plugin for use with OctoPrint. I can test it with OctoPrint. It is great that you are using my software with an entirely different software stack, but when you report an issue, please do not place the burden of trying to find out what is going on on me. I don't have your hardware, I can't reproduce what is going on. The only clues I have are hidden in your logs.

If I just get two screenshots, that gives me the feeling you are (only) saying "shit is broken, you figure it out" and that just pisses me off. With the number of projects you are involved in, I would expect you to be able to track down the difference between two branches. If I see you putting effort into the issue, chances increase that I put effort into fixing the issue. 

Closing the issue for now, because you are unable to reproduce it. Feel free to reopen if you can.

closed 2020-03-08 by fieldOfView

probonopd, 2020-03-08:
Sorry!