fieldOfView/Cura-OctoPrintPlugin #158: "Waiting for gcode analysis to complete" forever

fieldOfView, 2020-03-13:
Hello all!
With Cura 4.5 and Octo 1.4 seems to be broken again. I got the message "wait for octoprint to complete gcode analisys" and it goes forever. of course i can press the "print now" button .. in fact it's a minor annoyance but sometimes i find myself minutes later with the printer not moving and wondering why :)

_Originally posted by @sugar012 in https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/146#issuecomment-598340902_

fieldOfView, 2020-03-13:
>  I got the message "wait for octoprint to complete gcode analisys" and it goes forever

This dialog is shown if you have Print Time Genius installed. PTG does not work if you start the print before the gcode analysis is complete. Therefore the plugin probes OctoPrint to see if gcode analysis for the file has been completed before starting the print. It could be that there's a bug on my end, or perhaps the gcode analysis is never started after extracting the gcode from the ufp file. I'll try to find out.

*Originally posted by @fieldofview in https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/146#issuecomment-598470115*

The ufp plugin tries to force the analysis once the file is extracted, but not sure how that works with PTG or not.

*Originally posted by @jneilliii in https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/146#issuecomment-598481078*

fieldOfView, 2020-03-13:
A temporary solution would be to disable either the Ultimaker Format Package, or Print Time Genius. I'll see if I can work out what is going wrong with the combination of those two plugins and my Cura plugin.

fieldOfView, 2020-03-13:
I've tested this, and for me with both the Ultimaker Format Package and Print Time Genius plugins installed on OctoPrint, the gcode is succesfully analysed. A 192 MB gcode file took around 20 seconds to analyse on a Raspberry Pi 3, and printing started after it was done.

How large is the gcode file you uploaded?

fieldOfView, 2020-03-13:
I misread the numbers in my previous post; A 60 MB gcode file took about 2:50 minutes to analyse, 19 MB gcode took about 35 seconds.

sugar012, 2020-03-13:
> I've tested this, and for me with both the Ultimaker Format Package and Print Time Genius plugins installed on OctoPrint, the gcode is succesfully analysed. ~A 192 MB gcode file took around 20 seconds to analyse~ on a Raspberry Pi 3, and printing started after it was done.
> 
> @sugar012, how large is the gcode file you uploaded?

The issue started when i upgraded to octoprint 1.4. I've never seen that message before and it always started pinting right away.
I'm trying with a 90mb file and after the message (storing data on octoprint) i got nothing. Not even the gcode analisys message.
Trying with an 875kb file i got the message that goes on forever (past 5 minutes)
Tried clearing the PTG cache data but with no luck.
Disabling PTG does not have any effect either!
[image]

If you want log please tell me where they are located. Thanks

fieldOfView, 2020-03-13:
> I don't have the UFP plugin installed on octoprint

Ah, never mind my references to that plugin then; The issue you previously "hijacked" was specifically about that plugin, hence my confusion.

> The issue started when i upgraded to octoprint 1.4. I've never seen that message before and it always started pinting right away.

The "Waiting for OctoPrint to complete Gcode analysis" dialog itself does not have anything to do with OctoPrint 1.4, but was introduced with the current version of the OctoPrintPlugin. It is part of an effort to better support the Print Time Genius plugin on the OctoPrint side. If you remove the Print Time Genius plugin from OctoPrint, the dialog should not be showing anymore.

If you say printing also doesn't start without PTG, I guess there is something else going on. Please post a link to your Cura.log (see Help -> Show configuration folder), and we'll see from there.

fieldOfView, 2020-03-13:
What do you run OctoPrint on? If a Raspberry Pi, then what model?

sugar012, 2020-03-13:
> What do you run OctoPrint on? If a Raspberry Pi, then what model?

I'm using an Samsung ATIV 500 (on windows)
Here's the log
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4331071/cura.log)

fieldOfView, 2020-03-13:
Thanks for the log. I can see your problem happening, but not yet why.

Could you also post octoprint.log, from the Logging pane of the OctoPrint Settings in the OctoPrint web interface?

sugar012, 2020-03-14:
Here it is
[octoprint.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4332392/octoprint.log)

sugar012, 2020-03-14:
i uninstalled the DisplyETA plugin since i saw so many errors in the log and now it worked when i tried printing. Could have been the issue?

fieldOfView, 2020-03-14:
You beat me to it. Seems the issue has been fixed, which is great. Let me know if the problem resurfaces.

sugar012, 2020-03-14:
Not an issue anymore it seems! Thanks

Jakobud, 2020-04-19:
I don't have the DisplayETA plugin installed but I do have the PTG plugin installed. So is the only solution to fixing this bug is to disable the PTG plugin in Octoprint? Because it works just fine in Octoprint. Cura-OctoPrintPlugin seems to be the thing with the problem here...

I'm sitting here with a 270kb gcode file that the Cura-OctoPrintPlugin has been waiting on gcode analysis for about 10 minutes now...

Something is broken and needs to be fixed. Or at the VERY LEAST add an option to NOT do gcode analysis...

marekorok, 2020-04-26:
Hello,
I had the same issue with Cura waiting endlessly for OctoPrint (OP).

In my case issue was with Octoprint plugin PrintTimeGenius, which does gcode analysis at start and was failing every time blocking somehow Cura OP plugin, yet I was able to force clicking 'Print Now'.

Once I solved problems with OP plugin, print is starting without any waiting as expected.
At least that was my case, maybe help others.

Cheers.

fieldOfView, 2020-04-27:
@Jakobud, could you reproduce the problem and then share your logs from both Cura and OctoPrint please? Also, there is no need to shout over this.

@marekorok, how did you fix your PrintTimeGenius? Perhaps with that information I'll be able to *break* it on my installation, so I can test this issue and see if there is anything I can do to detect PrintTimeGenius not doing what I think it should be doing.

marekorok, 2020-04-27:
> @marekorok, how did you fix your PrintTimeGenius?

well, I moved my OctoPrint to python3 and PrintTimeGenius was failing to start analysis,
exactly like here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/188#issue-606390181

dirty fix chmod +x solved the problem.
Now I see that Cura is waiting for this analysis to finish, especially visible with large prints, but work now for me since then.

fieldOfView, 2020-05-08:
@Jakobud, it would really help me if you could post some logs from both OctoPrint and Cura. Without logs I can't fix your issue.

Jakobud, 2020-05-08:
For some reason it doesn't seem to be doing it now. I haven't made any updates or touched my 3d printer in a bit. PTG plugin still enabled in Octoprint. Today when testing I hit the "Print with Octoprint" button, then it comes up with "Sending Data to Octoprint" then "Storing Data on Octoprint" then shows the Save confirmation. I don't see anything about gcode analysis anymore. 

Kinda weird as I haven't done any sort of Cura updates or plugin updates, etc. Does Cura do automatic plugin updates in the background maybe?

Anyways if it happens again I will make sure to post logs. Sorry about "Also, there is no need to shout over this." I did not intend that at all. Just capitalized for emphasis! :) I enjoy your plugin.

fieldOfView, 2020-05-08:
> For some reason it doesn't seem to be doing it now

Ha, figures ;-) The plugin does no auto updating. From your description of what happens, it sounds like you have (now) turned off the option to automatically start the print after uploading (otherwise there would be no "Save confirmation"). If so, the plugin does not have to wait for Print Time Genius. That dialog only shows when the auto start print option enabled.

> I did not intend that at all. Just capitalized for emphasis! :)

For future reference placing emphasis can be done by putting * around text. \*This\* becomes *This*.

Have a nice weekend.

fdeguibert, 2020-05-10:
Hey there :)

I'm having the same issue. 

I will attach logs but, first, here is what I did try, if it can help

the issue started to happen a few days ago (maybe 2 days ago). Yesterday that was really anoying, so I tried a few things like disable the Genius plugin, clean its data, even uninstall it.
nothing really helped.

In the end, I just reinstalled Octoprint (v1.3) and everything worked just fine again.
Then I updated it to 1.4.0  and everything was still ok.

Then I reinstalled only 2 plugins: Bed Visualizer and Genius Time.

And the message "Waiting for octoprint to complete gcode analysis" started again (and stay for years... )

My test file is a 715.5KB gcode file, so not that big...

the thing is, i have uninstalled, again, genius, and cleaned its data. The problem still occurs...

So here are 2 lock packages. 
The first is after uninstalling genius, and trying to send a new gcode file from cura.

[first_withGeniusUninstalled.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4605957/first_withGeniusUninstalled.zip)

The second is after reinstalling genius, and trying to send the same gcode file from cura.
[Second_WithGeniusReinstalled.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4605980/Second_WithGeniusReinstalled.zip)

hope that can help.

this plugin is just great (well, on the 1.3 it worked like charm), hope it will be fixed ;)

thank you

Jakobud, 2020-05-10:
> turned off the option to automatically start the print after uploading

Where is this option?

fieldOfView, 2020-05-11:
@Jakobud: in the Connect to OctoPrint dialog in Cura.

@fdeguibert, when you uninstalled PrintTimeGenius, did you leave Cura open? Currently the plugin only asks the OctoPrint instance what plugins are installed once after a start of Cura. So if you had the PTG plugin enabled when Cura started, and then disabled the PTG plugin, Cura will think the PTG plugin is still enabled. Since there is no way to check if the PTG plugin is actively analysing, the plugin can only wait until it sees that analysis is done. But if PTG has been disabled after it was detected on start of Cura, the plugin will wait forever for a sign of PTG being done with analysis.

fdeguibert, 2020-05-11:
@fieldOfView thanks for this :)

That was it.

So I tried to disable PTG, then resarted my octoprint server, then restarted Cura. I still have the waiting for gcode analysis for ever.

Then I uninstalled PTG, restarted my octoprint server, then restarted Cura, and now it's all good as you can imagine. Except I don't have my beloved PTG anymore. 

But between PTG and Cura-OctoprintPlugin, I made my choice. so... that was good to meet you PTG, I hope I will see you again. 

Seriously, is there anything I can do to help? Now that I think about it, I didn't try to analyse with PTG a file directly uploaded on the OctoprintServer without using Cura. If analysis still doesn't work... I guess I'm in the wrong github issue repository ;)

looxonline, 2020-09-02:
I think I found out what is causing this and it's pretty simple. For some reason PTG turns off all of the analysis modes sometimes and that results in it sitting in an infinite loop while it tries to analyze using no tool at all. I have been having this issue for ages and never really looked into it until today when I found that all of the analysis tools were off and I never actually turned them off. Flicked the Marlin one back on and we were good to go.

fieldOfView, 2020-09-03:
Nice one, @looxonline. I'll add a check to see if any of the analysers is enabled.

fieldOfView, 2020-09-03:
I have added the check here: 012dad3

fieldOfView, 2020-09-16:
Fix is available in version 3.5.17 in the Marketplace

closed 2020-09-16 by fieldOfView

looxonline, 2020-09-16:
Awesome! Thanks always for the effort you put in.

TheShanMan, 2021-09-24:
Found another cause of the infinite "waiting for gcode analysis" in Cura... I created a new octopi instance from scratch and left Cura running. Then I tried to print from Cura and experienced this problem. After finding this github issue, I went through my plugins to see what I thought could be the issue since I don't have either of those currently, and didn't find it. Then I remembered that I left Cura opened and for kicks, I closed and reopened Cura. That fixed it... My old octopi instance had Print Time Genius installed. So Cura's running state apparently expected PTG to still be present. I imagine the same problem will occur if you disable or uninstall the plugin without restarting Cura. I don't know if there's a better way for the Cura plugin to handle this, but if not, at least this particular cause will be documented for anyone who comes here with the same condition.

newpain01, 2022-11-18:
This is still not resolved, I am getting the same error with Cura 5.2.1 and Octoprint 1.8.6
[image]

looxonline, 2022-11-18:
> This is still not resolved, I am getting the same issue with Cura 5.2.1 and Octoprint 1.8.6 and it doesn't go away unless I cancel.
> [image]

Do you have print time genius installed?

w7boz, 2022-12-15:
I am having the same issue. Disabled Print time genius also. Still not working.

fieldOfView, 2022-12-16:
Did you restart Cura after disabling Print Time Genius?