fieldOfView/Cura-OctoPrintPlugin #159: Can't store the printjob with the same name as the one that is currently printing

seebol, 2020-03-15:
I'm trying to print via the Cura v4.5.0 Octoprint plugin (as installed from the Cura Marketplace, 3/14/20). I have octoprint and cura both installed on my Ubuntu Linux computer which is connected to to the printer via USB. I do not have a Raspberry Pi. The Octoprint works fine by itself, and in Cura I can move the extruder or send gCode after connecting to the printer via octoprint. But if I try to slice a .stl and "Print with Octoprint", I get the following "OctoPrint error" pop up in Cura: "Can't store the printjob with the same name as the one that is currently printing." --renaming the .stl file hasn't helped get around this error.

fieldOfView, 2020-03-15:
Could you post a link to your octoprint.log (Accessible through OctoPrint Settings -> Logging on the OctoPrint web interface)?

The message is shown when OctoPrint returns a 409 HTTP status. The most common reason for that status is when you try to overwrite a printjob while it is printing. Apparently there are more scenarios for a 409 response.

fieldOfView, 2020-03-15:
Do you have the option to send gcode files to the printer SD card enabled? That is something I don't regularly test.

seebol, 2020-03-15:
Hi FieldOfView,

I've attached octoprint.log, and there are 409 responses [octoprint.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4334268/octoprint.log)

Regarding your second question, in the Octoprint web interface I have two blue buttons: Upload; Upload to SD. The latter button is grayed out and not clickable. However, under Octoprint Settings > Features > Enable SD Support is checked.

fieldOfView, 2020-03-15:
> However, under Octoprint Settings > Features > Enable SD Support is checked.

The problem should go away when you uncheck that option.

seebol, 2020-03-15:
Thank you for the help! That works as described.

fieldOfView, 2020-03-15:
For the record, you seem to have configured OctoPrint with the ability to store files on the sd card, but either you have no sd card in your printer or something else is going wrong with the sd card.

peelerj, 2020-05-02:
Starting in Cura 4.6 and in 4.6.1 I get the same error.  Cura 4.5 works great, no error.  I have the SD Card Checked in Octoprint.  Issue started in 4.6.

fieldOfView, 2020-05-02:
Could you tell me what version of the OctoPrint Connection plugin you have in Cura 4.5? You can see the version in the top right corner of the Connect to OctoPrint dialog (3.5.12 in my case):
[image]

I have marked this issue fixed because I made a fix locally, but that fix is not yet released to the Marketplace. You could help me test the fix by downloading this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlugin-v7.0.0-2020-05-01T10%3A28%3A25Z.curapackage?Expires=1903762978&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=bYQG%2BCw6oFRCR8X4jIqlH9JhWvIJziFD8YCSTAAmVcCJMVXGpLiFdRBS%2BWZPPLonL5HR9R001rTSnPb8N47EhlUGyWcaTbWqcZOs5m6VRw%2BgYfwhxwCOeU074TSiTZjiFrX1WeH7CDEe57oJNloeUw3GYIhVsogMRnRa3gbaPRZgzOBsV6IEj0DV2U911JFB%2FnAtFLiwWqjpLJ%2FXbqtsmilvBcdvTGE8vUz6RtwEhrTvY%2BxyEt1mcvHYAVneB%2FSqaFkkMt6XSsS8HKtbcWTd5BAiqqHgM5gKL9sBzu4YzfBUbcq7nXC5N8iW9OfDVjMnP59cH10zDLjEFanAdNcz%2FQ%3D%3D). After downloading the file, please drop it into the buildplate in Cura as if you were opening a 3d model.

fieldOfView, 2020-05-02:
If the development snapshot does not fix the problem, please upload your Cura.log file so I can have a look at what is going on. Can you upload to the printer sd card through the web interface?

peelerj, 2020-05-04:
Sorry...  i didn't see your reply. I will work on this evening..  thank you
for replying to my post!!!  I will let you know...

peelerj, 2020-05-05:
I have the same version as in your image above..  testing the fix you sent
now...

peelerj, 2020-05-05:
here's my curalog.

Yes it works just fine in Octoprint and in Cura 4.5

I also get this error when I click on the Octoprint Button after trying to
print from Cura to Octoprint...

Socket connection failed. Please check your browser's error console
<https://faq.octoprint.org/browser-console> and octoprint.log
<https://faq.octoprint.org/logs> for possible reasons.Also make sure that
the server is actually running by reloading this page.

fieldOfView, 2020-05-05:
> I also get this error when I click on the Octoprint Button after trying to print from Cura to Octoprint...

That means OctoPrint has crashed. I would need the logs from octoprint to see what happens, but this sounds like an issue with your OctoPrint installation. Please post a link to your octoprint.log file. You can download it from your OctoPrint installation via the `OctoPrint Settings` -> `Logging` page in the OctoPrint web interface.

Have you tried the development snapshot I linked to above?

fieldOfView, 2020-05-05:
> here's my curalog.

Unfortunately, you cannot send attachments to github via email; they are stripped. So if you want to attach files you must do so via the github web interface, or you can upload the file to dropbox, google drive or similar and post a link.

peelerj, 2020-05-05:
Yes.. did the same error..  i will get you they logs this evening.. thank
you so much!!!

fieldOfView, 2020-06-04:
Version 3.5.13 is now available in the Marketplace.

closed 2020-06-04 by fieldOfView

peelerj, 2020-06-04:
Sorry. Sir.. its working now..i think it was more of a Ubuntu update issue
personally...  i went through 2 updates of my Ubuntu..  it started
working.   I haven't been on using my system for a few weeks too... thanks
for your help and awesome software!