fieldOfView/Cura-OctoPrintPlugin #160: Minor Windows 10 Pop-Up Issue

jneilliii, 2020-03-18:
The windows 10 pop up shows html code as shown in the screenshot below.

[image]

fieldOfView, 2020-03-18:
I'll look in to it. The same message probably shows if you restore the Cura window.

jneilliii, 2020-03-18:
Cura side appears "normal".

[image]

fieldOfView, 2020-03-18:
Well, ideally that should also show something more helpful.

fieldOfView, 2020-03-18:
To reproduce this, you shut down OctoPrint while Cura was connected to it? Did you use a "discovered" instance, or did you use an instance that was Added manually in the Connect to OctoPrint dialog?

jneilliii, 2020-03-18:
It happens when I'm connected to a manually added instance and restart octoprint as a result of plugin install for example.

jneilliii, 2020-03-18:
I was able to catch a screenshot of it in cura...

[image]

fieldOfView, 2020-03-31:
Fixed in master with the above commit.

See de34fb1.

commit de34fb11 (2020-03-31): Handle OctoPrint going offline more elegantly

Fixes #160
closed 2020-06-04 by fieldOfView