fieldOfView/Cura-OctoPrintPlugin #161: Plugin gets stuck on "Waiting for Octoprint to connect to the printer".

ilovecats516, 2020-03-20:
Octoprint does successfully connect, but Cura gets stuck on "Waiting for Octoprint to connect to the printer".

fieldOfView, 2020-03-20:
Please provide as much information as you can, or I won't be able to tell what is going on.

I guess you use some sort of smart plug or another way to turn on your printer. Which one, and which OctoPrint plugin do you use to control it?

I guess you have your printer turned off (but OctoPrint up and running) before send the print to OctoPrint from inside Cura. Does the printer turn on? After being turned on, does OctoPrint automatically connect to the printer?

When the "Waiting for OctoPrint to connect to the printer" message is shown, does the button to "queue" the print work to upload the print?

What OS do you run Cura on? What version of Cura do you use?

ilovecats516, 2020-03-20:
Yes, I use a smart plug, it’s a Wemo plug. Plugin: https://github.com/jneilliii/OctoPrint-WemoSwitch

The switch is turned on and Octoprint does connect automatically.

The queue button does upload the print but doesn’t start it.

Cura runs on my Windows 10 machine, version 4.5.

fieldOfView, 2020-03-20:
Can you post a link to your cura.log? It can be found through Help -> Show configuration folder. It is also helpful to have the octoprint.log from the OctoPrint side of things. You can download that via OctoPrint Settings ->OctoPrint -> Logging

fieldOfView, 2020-03-20:
I don't have a Wemo plug, so I can not test the interoperability between the plugin for that plug and Cura. Let's hope your logs tell me something.

ilovecats516, 2020-03-20:
[octoprint.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4362080/octoprint.log)
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/4362089/cura.log)

fieldOfView, 2020-03-27:
Thanks for the logs. Unfortunately, they don't show a lot of detail.

> The switch is turned on and Octoprint does connect automatically.

Can you clarify this? Was the switch turned on before you tried to send the print from Cura to OctoPrint? Or was the switch turned on by Cura (via OctoPrint) as you started sending the print from Cura?

If the switch was already on before you hit the button on Cura, I might have been thinking about the issue wrong, and it is probably a duplicate of #136, where Cura is somehow no longer updating the state of the OctoPrint instance. If that happens, Cura could think the printer is not connected even though it is, and it will try to turn on the wemo switch and wait endlessly until the printer gets connected.

aidenmitchell, 2020-03-27:
Cura turned the switch on via Octoprint.

fieldOfView, 2020-03-27:
Sorry to keep asking for clarification, but I'm still trying to find out what does and what does not work.

* The smart-switch was turned off, and the printer was not "operational" in OctoPrint (because it was off)
* You sliced a file and pressed the "Print with OctoPrint" button
* The printer turned on
* The "Waiting for OctoPrint to connect to the printer" dialog showed in Cura and did not go away

Am I correct sofar?

Was the printer "operational" (connected and ready) in the OctoPrint web interface or was it not?

aidenmitchell, 2020-03-27:
No problem. You are correct so far. The printer was operational in Octoprint.

fieldOfView, 2020-03-27:
Thanks. This is a mystery to me. Did it happen just once, or is it reproducible?

aidenmitchell, 2020-03-27:
I tried multiple times.

fieldOfView, 2020-03-27:
And what happened?

aidenmitchell, 2020-03-27:
Same thing, the printer turns on and Octoprint connects, but Cura gets stuck on "Waiting for Octoprint to connect to the printer".

7kasper, 2020-06-08:
I am getting the same issue. Running on the latest master branch.
It is weird as the system worked fine before... Suddenly have this issue. 
I use PSU controll with the PI turning on a relais that I made.
Printer turns on fine and connects. Then cura gets stuck at connecting. Like said, this worked in previous versions.
Even with the printer turned on and connected the cura instance gets stuck on connecting.
Sometimes after restarting cura the issue is gone though. Makes it hard to trace.
https://pastebin.com/592m1CfR

fieldOfView, 2020-06-10:
@7kasper please elaborate a little. Is your printer turned on and connected before you try to send the printjob to OctoPrint, or do you let Cura turn on the printer?

alaorath, 2020-06-13:
I was searching Google for an answer and found this post.

I have the exact same setup as the original poster: OctoPrint with Wemo & PortListener plugins (so the printer powers on, OctoPrint connects) Cura correctly powers on the Wemo, and I can see OctoPrint connect, but Cura remains in "waiting for Octoprint to connect".

(while writing this comment, I thought to check for updates to the plugin in the Marketplace...)
After updating OctoPrint Connection to v3.5.14 I can no longer reproduce the issue...! It does exactly as previous versions of OctoPrint... Cura powers on the Wemo switch, waits for OctoPrint to connect, then starts printing.

I think 3.5.14 fixed the it!  :)

7kasper, 2020-06-14:
> @7kasper please elaborate a little. Is your printer turned on and connected before you try to send the printjob to OctoPrint, or do you let Cura turn on the printer?

I have tried both and it failed in both instances. In all cases where the printer was off before, it did turn on just bot connect. The weird thing is.. I have had exactly the issue described but like said after 2 restarts of Cura (not octoprint) it suddenly worked and hasn't stopped working since. Both on printer off start and printer already on start.

fieldOfView, 2020-06-14:
I'm starting to think that this issue (too) is a duplicate of https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/136, where Cura stops seeing changes in the updates it requests from OctoPrint.

Wpnx330, 2020-09-17:
Hello,

I also have this issue. However I do not use a smart plug.

Printer : Sidewinder X1 v4 (Powered ON Always)
Hardware : Rasperry Pi 4B rev 1.2
OS : Octopi 0.17.0 
Octoprint : 1.4.2
Octoprint Connection : 3.5.17 (Cura Plugin)

When in Cura printer settings, I can connect to my printer via OctoPrint.
When trying to print I can successfully send to queue, and then manually print via OctoPrint
I can upload from my desktop to OctoPrint and manually print without issues.

When I try to print from Cura it gets stuck with the message "waiting for Octoprint to connect to the printer."
 - I have tried to perform the print while Octoprint was connected to the printer. 
 - I have tried to perform the print while Octoprint was NOT connected to the printer. (so that Cura can ask it to connect)
 - I have tried changing the setting which says "Connect to printer before sending printjob," then re-running the two tests above.
 - I performed each of these 4 tests after a power cycle of my Raspberry pi and my X1 printer.

akiltopu, 2020-11-11:
I have a similar issue. 

I am using Ender 3 v2 printer with octoprint on macosx catalina.

I am using cura 4.8 with octoprint connection plugin (latest version) and wemo switch plugin.

When my printer connection is offline and powered down; with the help of octoprint connection plugin and wemo switch i could power up my printer but serial connection is not working.

Error:
State: Offline (Error: SerialException: 'read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2916)

When i close cura i can connect to my printer from octoprint web console easily.

I could not connect it from octoprint connection plugin if it is disconnected from octoprint web console.

alexose, 2021-01-04:
Having the same issue here.  Cura 4.8 + Catalina.

alexose, 2021-01-04:
Works after downgrading to Cura 4.6.2.

psiess, 2021-02-26:
I'm a bit late to the game but I'm also having this issue.  I have an Ender 5+ that takes about 15 seconds to load after power on.  The Cura plugin turns on the printer via PSU control plugin and then tries to connect right away.  No connection is made because the printer isn't initialized and Octoprint is not connected to the printer, so the Cura plugin gets stuck trying to connect and never completes.

mw75, 2021-03-02:
Had the same problem. Fixed it by diconnecting and connecting under "manage printers" -> octoprint, wait for key validation -> disconnect -> wait for key validation again -> connect.

Hobbit44, 2021-06-14:
Hi, adding my 2p to this. 

My setup:
Ender 3 V2
TPLink KP105 Plug
Octoprint 1.6.1
Cura 4.9.1

What i have noticed is:
1.
settings:
 - auto connect off
 - auto disconnect off
socket off
printer connected
Cura does not try and turn the socket on. Printer gets a `SerialException: device reports readiness to read but returned no data` error.

2.
settings:
- auto connect on 10s
- auto dissconnect on 0s
socket off
printer dissconnected
Cura will turn the printer on and everything behaves as expected.

My preference would be to have the status remain as operational. Could this plugin check the status of the plug and act on that regardless of what state the printer is in? Im assuming it might be checking if the printer is operational and skipping the plug action.

petr7555, 2021-08-04:
I have had the same issue, restart of Cura helped.

looolz, 2023-01-26:
> I have had the same issue, restart of Cura helped.

Same. I just had to restart Cura.