fieldOfView/Cura-OctoPrintPlugin #162: Automatically connect or prompt if you would like to connect

meme8383, 2020-03-25:
It would be nice if Cura would automatically connect to the printer or ask me if I would like to connect because I would like to be able to turn on my printer and click "print" instead of waiting for the octoprint website to open and clicking connect.

fieldOfView, 2020-03-25:
This should not be too hard to implement after the smart power plug support already paved the way. Note to self: https://docs.octoprint.org/en/master/api/connection.html#issue-a-connection-command

fieldOfView, 2020-04-03:
See https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/24c17ab39650d533cad0ac33ec3fee13b3ea99c0

fieldOfView, 2020-06-04:
Version 3.5.13 is now available in the Marketplace.

closed 2020-06-04 by fieldOfView