fieldOfView/Cura-OctoPrintPlugin #163: Unstable camera connection

lukastrimmel, 2020-03-25:
I am using Cura 4.5 and the latest version of the OctoPrintPlugin v3.5.12. 
From the beginning i had troubles connecting the camera with octoprint, although the connection is working fine via the webbrowser, it strugles displaying the picture in the latest version of Cura (4.4 was also not working). 

https://drive.google.com/drive/folders/18zWAgBMw9fgJXVR4-CvwhK8BEK9y2oad?usp=sharing
here you can find my cura.log and 2 test videos showing the problem. 

In the cura.log i could see that this line keeps poping up all the time:
`[MainThread] OctoPrintPlugin.NetworkMJPGImage._onStreamDownloadProgress [154]: MJPEG buffer exceeds reasonable size. Restarting stream...`

I thought maybe the problem was that i had an webcam resolution of 1280x720 and 30fps and tried to reduce the quality to 640x480 and 10fps - unfortunately the problem persists.

fieldOfView, 2020-03-25:
When you changed the resolution of the webcam, did you check if those changes "took"? IE: was the image in your browser actually smaller?

What webcam do you use (brand, type)? Can you try a different one?

An mjpeg stream is a quite inefficient video format. Essentially, it sends 30 jpeg images of 1280x720 pixels per second, all concatenated into a neverending file. Unfortunately, Cura uses a rather inefficient way to display this stream; it looks through the neverending file for the beginning of a JPG image, then continues to try to find the end of the JPG, trims that from the stream and sends it to the display component. One frame at a time.

What seems to be happening in your case is that Cura can't find the end "marker" of some frames. In some cameras, the whole mjpeg stream is created on the camera and simply piped to the network. It could be that the specific camera you use is sending the mjpeg stream in an unexpected format. It could also be that the mjpeg stream is using very little compression, so each "frame" (jpeg image) is "extremely" large. The code was originally intended for 640x480 streams, so the limit of what is "unreasonably large" per frame is currently around 2 MB (and I would maintain that is indeed unreasonable, because >2MB at 30 FPS would be a >480 megabit stream of video, which is more than the Pi can transport over its USB-based nic).

lukastrimmel, 2020-03-25:
Thank you for your quick explination. Yes the resolution of the webcam was actually smaller in the browser and in cura, so i am totaly sure that the changes to 640x480 10fps did work. 

i am using a no-name webcam from amazon: https://www.amazon.de/gp/product/B01FDCK6FS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Unfortunately i do not have an alternative camera, but as you described it may just be caused by an incompatible camera.

fieldOfView, 2020-03-25:
I use Logitech C270 cameras, but I don't think they match your criteria for "cheap".

fieldOfView, 2020-03-25:
In the mean time, it would help if you could record a part of the mjpeg stream so I can have a look what is going wrong.

Download curl for windows from https://curl.haxx.se/download.html
Use curl to download a fragment from your webcam:

`[path to curl]\curl.exe http://192.168.0.15:80/webcam/?action=stream --output c:\users\[you]\desktop\stream.mjpeg`

Wait until that command has saved around 8000k and press control-C
Upload the resulting file somewhere I can have a look.

lukastrimmel, 2020-03-25:
i recorded about 20 seconds, the file can be found in my drive:
https://drive.google.com/drive/folders/18zWAgBMw9fgJXVR4-CvwhK8BEK9y2oad?usp=sharing

Thanks for the recommendation, the price criteria of the C270 would match my expectations.

lukastrimmel, 2020-04-18:
After buying another camera from logitech, everything is working fine now. 
So some cameras are not "supported" by cura, as already supposed.

fieldOfView, 2020-04-18:
Thanks for testing this. I may have to buy me a camera like the one you used before, to see what is going on.