fieldOfView/Cura-OctoPrintPlugin #164: Domoticz Power, working but doesn't initiate Connect

Wob76, 2020-03-26:
Running Cura 4.5 and the Connect to OctoPrint 3.5.12

I am linking in to the OctoPrint-Domoticz plugin and the power option is populated and working, the issue I am seeing is that it doesn't initiate a connect after the power up. I have the Domoticz plugin configured to "Auto Connect" and GCODE on Delay, I have also tried adjusting the delays without change. The plugin does work as expected when using the Octoprint plugin Power Icon, or inserted start GCODE.

I wasn't sure if to open an issue here or with the OctoPrint-Domoticz plugin, but since it works as expected from the Octoprint Interface and GCODE I thought it best placed here.

Wob76, 2020-03-26:
Sorry, after having a better look I see #162 is a similar request, please feel free to link and close this one.

fieldOfView, 2020-03-26:
Does OctoPrint initiate the connection if you turn on the power from within the OctoPrint web interface?

fieldOfView, 2020-03-26:
@jneilliii, could it be that autoConnect works in the Domoticz plugin when the plug is turned on via the button in the interface (ie: from js) but not when turning the plug on through the API (ie: from python)?

fieldOfView, 2020-03-26:
Looking at the code of the Domoticz plugin, it looks like that plugin completely ignores its own autoConnect setting when turning the plug on through the API. I would say this is a bug with that plugin.

closed 2020-03-26 by fieldOfView

Wob76, 2020-03-26:
Thank you for looking so quickly, I'll raise a ticket over there.