fieldOfView/Cura-OctoPrintPlugin #165: [BUG] Cura resets the API key every time it is closed

kulfuerst, 2020-03-27:
Since the last update, Cura always resets the API key. This means that I have to fetch the key and reenter it every time I start Cura.

Please let me know what info you need for troubleshooting. I'll happily collect it for you.

Thank you for your support!

jlmeredith, 2020-04-02:
I am experiencing something similar.  I am unable to print from Cura because it keeps saying the key is invalid in Octoprint.

fieldOfView, 2020-04-03:
> This means that I have to fetch the key and reenter it every time I start Cura.

When you do, do you press the "Connect" button? That might be critical to store the API key (though I don't know why it would be necessary to re-enter it after installing "the last update" (BTW, was that the last update of Cura, or the last update of the OctoPrint plugin? Or even the last update of OctoPrint?).

fieldOfView, 2020-04-03:
>  I am unable to print from Cura because it keeps saying the key is invalid in Octoprint.

Have you tried re-entering (and/or requesting) the API key and pressing the "Connect" button? Or is there no Connect button, only a Disconnect button? In that case, try disconnecting using that button, then entering or requesting the API key, and then pressing the "Connect" button.

mikekscholz, 2020-04-13:
This was happening to me as well. I got a 403 error every time I open cura, paste in the new API key, and it would come back as validated and everything would work, exit and restart cura and the API key would be reverted to what it was before I pasted over it. After pasting it again I clicked on disconnect then immediately on connect again, then exited out again and it finally wrote the the new key to disk.

fieldOfView, 2020-04-13:
Thanks for confirming that workaround. Now I know what to fix.

mikekscholz, 2020-04-13:
as an aside, not sure if its related, but when I clicked request it would open into my octoprint in browser and ask me for permission for the request, but after confirming it never brought the key into cura.

sylvainle, 2020-04-19:
First, check that cura can talk to octoprint in firewall rules

Then, try in steps :
1- (octoprint) disable camera
2- (octoprint) connect to 3d printer
3- Launch cura
4- (cura) Setup your printer (get api key)
5- Quit cura
6- Launch cura
7- Check if temperature is displayed

Since many versions, I have to lauch octoprint and connect to printer in cure BEFORE launch cura

Hope it help

fieldOfView, 2020-04-28:
Could you try this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlugin-v7.0.0-2020-04-28T07%3A48%3A15Z.curapackage?Expires=1903420099&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=XIII01%2F3RtxbRBdT17U0LLu2nUCuuOn3vsALdlLWcf%2BKoh92L%2B1B1v99Pu%2FEhx4M9doNxPDJVJC3d4TVYjyJMKHcqlhOR72csAw27E6jrDslXUew1dNw7qyWyggCSKOw42ZrjmLcLn1Se6OYoNtNRmbClekHXL8PSlAWtq7DrAEB7OPdjrkFhMM%2BJpP2G2mpBH92oKL%2F%2BHNpBZmJ2MWd6A5o84Eefx7iPJ2%2B%2F9cD2TBDY%2FOSDO%2B2vahOEj6ztYXHhMk3xTTEWDp%2Bd%2Bo0B8HKGJxxs56ykRwe55wKRFN8p8OkJ%2B2TPdmMFOE%2FvdDOh5e6bVjCxPhf%2BmNpCoz80tplDQ%3D%3D)?
Download the file and drop it into Cura as if you are opening a 3d model.

fieldOfView, 2020-05-13:
@jlmeredith, @kulfuerst and @mikekscholz, I could really use your help testing the [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlugin-v7.0.0-2020-05-01T10%3A28%3A25Z.curapackage?Expires=1904712906&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=KqwPyqExR3eHg0UhFdYcuxwgpwrkhHSDRXgLSLCcBSRbQGZGM3HoCAgZa4dZwZLSWhCEYyg5MzfK0hOcWozmHY8PA6hUgTcK3UJPRbCSbn8dS2KRrKLUA%2BzZqw5nO6Z%2F34tp0Uj5JxVi54JUyJjriPuuABiHKjkzDKsnPBOd2auddSdjaU2VBSz6C471n53hAInK9SIicrJGIDDfXaKuS7TfywQEp20v3ORq%2Fl%2FlBFqcGQpBZTst5P6bTxVljo0X7m%2FusHjNpfKIfFEAcyqH9j8%2F1mJ4%2FtSa3y80VznBD6FGj4bAOKo258%2Bj8iY0oAVylCRU9MQBYG1WdCoQvxeb5A%3D%3D). Download the file and drop it onto the Cura buildplate as if you are opening a 3d model.

fieldOfView, 2020-06-04:
Version 3.5.13 is now available in the Marketplace.

closed 2020-06-04 by fieldOfView