fieldOfView/Cura-OctoPrintPlugin #166: loss of functionality, copy to sd with auto start

devildant, 2020-03-31:
hello,
I use the copy option on the SD and auto start of the print, it worked very well in 3.5.2 but I just did the update and it no longer works

best regards

devildant, 2020-03-31:
[image]

devildant, 2020-03-31:
```
2020-03-31 21:57:46,278 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1057]: Resource created on OctoPrint instance: http://192.168.1.69/api/files/sdcard/xyzcal~6.gco
2020-03-31 21:57:46,320 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1001]: OctoPrintOutputDevice got an error while accessing http://192.168.1.69:80/api/files/sdcard/xyzcal~6.gco
2020-03-31 21:57:46,321 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1002]: File not found on 'sdcard': xyzcal~6.gco
```

devildant, 2020-03-31:
from what I understand, the file selection attempt and the launch of the print is done before ocotprint have finished are streaming

fieldOfView, 2020-04-03:
I don't test printing from the printer SD card often, and I can see how some recent changes may have affected it. I will look in to this and try to fix it for the next version.

fieldOfView, 2020-04-28:
Could you try this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/czkOsRFb1V1tt9LBsZ8d7i7ThU3ACKWgI7OPOr0PcOG6/OctoPrintPlugin-v7.0.0-2020-04-28T15%3A54%3A31Z.curapackage?Expires=1903449274&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=joyizAmKEQIJyp3ug6cp%2B%2BdjG3I%2BZihgdYj6999c3r3Rn2gA7WUUHEzON3iS8A%2F9VN77sJ3RiQUyF7aEDrW%2FFhBDn3esQYStWCyCWzwcS4GTFTRAZiwKymJuys8JfsbBU%2Fzg8MzBiqLc%2Fu60EOkj%2BdAErryIQYzhyh7ZfKZ3IY%2BV91Tm3tFAkbKCQY20jJ91Z3W3Nx5F5z28i3vz54EWu%2FkdclAVBxuDosnoqCiaGrKjkZeAZQFglOK3L4TUQwokB0UQH2iZGM7nH6BY%2BDcKtfSbkIyOBiIhIk6qWutKx0f7MsiFC4HiMopzRjvvJFZh5gJCGl8uAaVZ81zyq4CC%2Bw%3D%3D)? Download the file and drop it into Cura as if you are opening a 3d model.

devildant, 2020-04-28:
It works well :) thx

fieldOfView, 2020-04-28:
Thanks for the quick test!

fieldOfView, 2020-04-28:
I'm keeping this open until there is an official release.